betaflight/unified-targets #11: Flashing Firmware Bricks CL Racing F7

azixaka, 2019-09-17:
Trying to flash any 4.1 RC3/Dev firmware via Configurator 10.60 RC1 bricks the CL Racing F7 at "flashing..." stage.

Boot mode/full erase/no boot sequence doesn't hangs at flashing stage as well. The erasing stage completes though.

Connecting via Configurator 10.51 and doing a boot mode restore works fine. Flashing any other version succeeds as well.

There seems to be a bug introduced in 10.60.

mikeller, 2019-09-17:
@IvoFPV: Lol, that was added by Cheng Lin (@bnn1044) himself in betaflight/betaflight#7779. 
Do you want to open a pull request for a fix?

closed 2019-09-17 by mikeller

azixaka, 2019-09-25:
It seems to get fixed now, I flashed the latest dev build yesterday and the unified target (non legacy) worked fine.

Thank you.

PR #12 (2019-09-17): Fix CLRACING F7 unified target to F722
Fixes #11