betaflight/unified-targets #15: Upgrading to 4.1 RC 3 today "bricks" Holybro Kakute F7 HDV

5zero7rc, 2019-09-21:
On September 15, I upgraded my Kakute F7 HDV from the default 4.0.4 to 4.1 RC3.  Everything worked fine all week.  I then tried a different firmware and tried to re-install 4.1 RC3 on September 20, and after it finishes flashing, the board is "dead".  You can not talk to it over USB at all. The green power led works, but the blue flashy light never comes on.  It does not show up on my mac as a serial device.  If I use the boot button to get back to DFU mode, I can flash any version of 4.0 and it works again, but as soon as I try 4.1 it dies.  Is this because of a change to the KakuteF7 unified target?  If so, can that be reverted?  I don't believe this board has an OSD.  I see there were changes to enable the OSD in the unified target.  Could this be the cause?

mikeller, 2019-09-21:
@5zero7rc: This was fixed inbetaflight/betaflight#8914 which will be included in 4.1.0-RC4.

McGiverGim, 2019-10-17:
@mikeller I close this because we don't have the stale bot here. Maybe is a good idea to add it here too.

closed 2019-10-17 by McGiverGim

mikeller, 2019-10-17:
@McGiverGim: See #66.