betaflight/unified-targets #81: Manufacturer ID for HIFIONRC

atgfpeyv, 2019-10-20:
As per https://github.com/betaflight/betaflight/blob/master/docs/TargetMaintenance/CreatingAUnifiedTarget.md, please assign manufacturer ID for HIFIONRC.

Requested Id : HIFI
Name : HIFIONRC
URL : (under construction, meet soon...)

A target definition of HIFIONRC will follow soon.

mikeller, 2019-10-21:
Please update this when the URL is available, a manufacturer id will be assigned then.

atgfpeyv, 2019-10-26:
@mikeller Thank you for your reminder, but our website is still under construciton. This is a domain name: http://www.hifionrc.com/. It should be open now. We are taking the time to perfect it.

mikeller, 2019-10-27:
The website still looks a bit funny for a website offering RC related gear:

[image]

closed 2019-10-27 by mikeller

atgfpeyv, 2019-10-27:
@mikeller Yeah. The website is still under construction.

PR #97 (2019-10-27): Added manufacturer id for HIFIONRC.
Fixes #81. 

@atgfpeyv: I assume you will also be the maintainer of HIFIONRC's target configurations?