betaflight/unified-targets #98: Custom config needed for a stack with different gyro alignment?

johnelliott, 2019-10-27:
I got a new FC from flight that took some custom config:

```# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: EXUA   board_name: EXF722DUAL   custom defaults: YES
```

iFlight f7 twin-g for DJI air unit. Not the same as the other F7 twin-g stacks. 

When I tried to fly, it freaked out predictable and cyclically because one gyro was 90º off. I don't think the same custom defaults can apply for these two stacks (I have both).

[image]

I think iFlight needs to add a custom config for the board with different sensor alignment. I could be totally wrong here. This is my first issue to unified configs.

johnelliott, 2019-10-27:
@Linjieqiang perhaps you can help out here since Mike added you to as a maintainer :)

Docteh, 2019-10-29:
Have you tried [IFF7_TWIN_G](https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFF7_TWIN_G.config)?

https://github.com/betaflight/unified-targets/blob/f05568005adf44646588ace8dbabb75b7c6adbf1/configs/default/IFRC-IFF7_TWIN_G.config#L132-L136

johnelliott, 2019-10-30:
I will double check, but the point I made above is that the same config does not work between the two boards.

mikeller, 2019-10-30:
@johnelliott: From what you are stating above, this seems to be different hardware, and (at least) have a different gyro orientation. You could try to address this by manually setting the board orientation, but in the long run this should probably have a dedicated target provided by the manufacturer.

johnelliott, 2019-10-31:
Thanks @mikeller, this is different hardware that's receiving incorrect custom config.

The screenshot posted above has the corrected configuration and the quad flies perfectly after determining what the sensor alignment actually is.

I can just make a PR for this, probably Friday night or this weekend once I have time to read docs. I haven't looked into how custom configs are mapped to boards but this must be a simple case.

PR #101 (2019-11-05): Add IFRC-IFF7_TWIN_G_D
Add iflight SucceX-D Mini F7

Adds support for the [iFlight SucceX-D Mini F7 board](https://www.iflight-rc.com/index.php?route=product/product&path=86_87&product_id=1060) which has different gyro alignment than the non-D board.

Fixes #98

@Linjieqiang here is my proposal to fix the issue with different hardware. Would you please confirm that this is different hardware?
closed 2019-11-05 by mikeller