betaflight/unified-targets #100: Pin C09 is assigned twice in DIAT-FURYF4OSD.config

SantiCO19, 2019-10-30:
Very new to this, so I'm not sure if this is an issue or not.

In the unified target config for Diatone FURYF4OSD, near the top of the config, there is resource assignment PPM 1 to C09, to which a note is displayed in the cli, "NOTE: C09 already assigned to ESCSERIAL." Then, further down the config, resource assignment ESCSERIAL 1 to C09 is made, to which a cli note is output, "NOTE: C09 already assigned to PPM." So it would seem that ultimately, ESCSERIAL is more or less unaffected as it remains assigned to C09, but whatever the intention with PPM is, it appears it ultimately lacks an assignment since it gets bumped off of C09. If this is intended, please pardon me for posting about it, while I continue doing my best to learn my way around here (betaflight, and Github.) Thank you.

jflyper, 2019-10-30:
A bit more detail.

Since escserial is used to provide bit timing for esc serial protocol (which means that this function does not need a pin) and esc serial protocol is only used during esc pass through, it is pretty much safe to overload with pins that are not used during esc pass through. Receiver pins are favored as they are guaranteed to be free.

SantiCO19, 2019-11-22:
Thank you jflyper. While I dont completely understand yet, I can make a little sense of what you describe. For me to fully grasp is only a matter of time and exposure. Cheers-

closed 2019-11-22 by SantiCO19