betaflight/unified-targets #223: Manufacturer ID for FOSS

cvetaevvitaliy, 2020-03-21:
Manufacturer ID for FOSS
Requested Id : FOSS
Name : DRONIUSF7
URL : none

cvetaevvitaliy, 2020-03-21:
Manufacturer ID for FOSS
Requested Id : FOSS
Name : FOSS
URL : none

etracer65, 2020-03-21:
Manufacturer id already exists. Please read: 

https://github.com/betaflight/unified-targets/blob/master/Manufacturers.md

closed 2020-03-21 by etracer65

cvetaevvitaliy, 2020-03-21:
@etracer65  I ask this lssues 
[image]

Asizon, 2020-03-21:
lol but is foss your brand name @cvetaevvitaliy ?

mikeller, 2020-03-21:
@Asizon: Manufacturer ids are not only assigned for brands, have a look at https://github.com/betaflight/unified-targets/blob/master/Manufacturers.md.