betaflight/unified-targets #344: Add a new manufacturer id for the unified target database

lan-rc2020, 2020-09-01:
Hello
i'm requesting a new manufacturer id for the unified target database
Manufacturer ID :LANR
Company name :LAN-RC
Our website is www.lan-rc.com
My email is: 2423847452@qq.com
Thank you!

mikeller, 2020-09-01:
Duplucate of #336.

closed 2020-09-01 by mikeller

lan-rc2020, 2020-09-01:
However, my manufacturer ID has not been added successfully. What should I do?

 
Closed #344.

mikeller, 2020-09-01:
@lan-rc2020: Why did you close your originia issue then? If you close it the assumption is that you are no longer interested. If you are still interested then reopen it and wait for it to be worked on.

lan-rc2020, 2020-09-02:
My  issue was opened 12 days ago, but it was not approved, so I thought there was something wrong with it. Yesterday my boss asked me to change my ID number from LNRC to LANR, so I reapplied. I just saw your email, I reopened the request, thank you!

lan-rc2020, 2020-09-07:
hello Michael:
                          Can you help me register the manufacturer ID as soon as possible,I have been waiting for half a month. 
                          Thank you very much.我
经等了有半个月了，非常感谢