betaflight/unified-targets #398: External barometer not visible

maxgame91, 2020-11-23:
hello, just bought an external barometer [here](https://www.amazon.fr/HiLetgo-BMP280-3-3-Precision-Atmospheric-Navigation/dp/B07ZWV68M6/ref=sr_1_6?__mk_fr_FR=ÅMÅŽÕÑ&dchild=1&keywords=BMP280&qid=1606070229&sr=8-6), and betaflight do not show on the top right section (it's activate in the configuration menu.
So it's a BME280

i have an F4 Mamba so the target is FURYF4OSD
I solder the SDA to the SDA and the SCL to the SCL
On betaflight 4.2.4
No I2C errors. on the status command.

Do you have any idea why

If you need any more information just tell me.

Thanks

mikeller, 2020-11-23:
@kc10kevin: Can you please provide the correct solution here?

maxgame91, 2020-11-23:
Hi guys, fisrt thanks all for your support in my problem,

@mikeller Thanks for the clarification, but to be fair, I take the risk because it's mandatory in my country for professionnal use to have an altitude calculate by a barometer. but I understand your statement about the wishes of diatone.

So about what you says, I just need to modified the [line 87](https://github.com/betaflight/unified-targets/blob/8d68f6c8ed65e366d05ac59d9a49b10912ac3301/configs/default/DIAT-FURYF4OSD.config#L87) to have my baro in this target ? BTW I take FULL responsability of my action.

@etracer65 Thanks for the cli line, already tried but look lite is more a problem of the target

kc10kevin, 2020-11-23:
Thanks @mikeller.  I'll try to stay out of the argument, but suffice it to say that I have tried to talk to Diatone several times in the past about this issue.   I2C was and is on my FURYF4 design when Diatone took it over.  They did not concentrate on I2C/Baro/Mags back then so it was not a priority.  Since then, they have reconsidered and are adding Baro's to the designs (MAMBAF4US, 722, etc) and we have updated/added I2C Configs to the Unified Targets list.

Now, saying that, when the Unified targets were created long ago, I specifically remember that we were told that they should be as generic as possible.  So, I was told to change MAG and BARO from AUTO to NONE since there was not a baro or mag built into the boards being sold by Diatone.   That is why they are NONE, but the FURYF4OSD can support an external I2C BARO/MAG on the pads (PB6/7).  If a user wanted to add one to I2C, then they would have to change the config file or turn it on in Betaflight configurator.   I'm more than happy (and so is Diatone) to change them to AUTO in the Unified targets.   

@mikeller - I've searched for all the relevant BME280 issues.   Is the baro supported?  I found issues and requests, but didn't seem to find a full resolutions.   Does it just use the BMP280 driver?

@maxgame91 As stated above, the target does support an external Baro via I2C 1 (PB6/7).   Now, I do not know what board you specifically have since I believe Diatone have different versions of the FURYF4OSD/MAMBAF4.   Setting AUTO in the config or turning on Baro in Betaflight configurator should work.   I believe the target is working as designed (except AUTO) for an external Baro if connected correctly.   I would recommend you contact Diatone customer support to see if there an issue with the board.

mikeller, 2020-11-23:
@kc10kevin:

> So, I was told to change MAG and BARO from AUTO to NONE since there was not a baro or mag built into the boards being sold by Diatone. That is why they are NONE, but the FURYF4OSD can support an [...]

No, this is not how this works.
Explicitly setting `baro_hardware = NONE` means 'no baro supported', and if it is Diatone's idea that they can use volunteers from Betaflight (who could be helping users with actual problems instead in the time that Diatone is now consuming) to help users enable baro despite them explicitly disabling it in their designs then they are wrong.
If they'd left this parameter at the default value we would not even have this discussion as the baro would have worked out of the box for @maxgame91.

So if it should be possible to use these boards with a baro / mag in Betaflight then please change the target definition to enable auto detection instead of explicitly disabling this.

mikeller, 2020-11-23:
@kc10kevin: Also yes, good point, the change to recognise the BME289 as a BMP280 (https://github.com/betaflight/betaflight/pull/9656) was only added to 4.3, so @maxgame91 will have to use a development version of the firmware to get this to work.

kc10kevin, 2020-11-23:
> No, this is not how this works.
> Explicitly setting `baro_hardware = NONE` means 'no baro supported', and if it is Diatone's idea that they can use volunteers from Betaflight (who could be helping [...]

@mikeller Thank you for clarifying that.  That was not Diatone's intent.  There is nothing nefarious going on and to imply such is not productive.  I'm just telling you what happened when Unified Targets were created in the first place.   I was told to change AUTO to NONE.  That is a fact and I cannot change that.   It is not Diatone's intent (nor mine) to use volunteers to help users with this kind of problem.  

I try my best to ensure that the Diatone targets are up to date, current and working 100%.  Not all of us know all the intricacies of Betaflight and have other commitments in life, so we do make mistakes.   I learn from them and try not to repeat them.  

Again, thank you for clarifying the issue and I will submit the change to all the Diatone unified Targets to ensure that the Baro and Mag are supported and work out of the box.   At least for the supported Baros/Mags.

mikeller, 2020-11-23:
Thanks for fixing this @kc10kevin - unfortunately even if this was not intentional, this is what has been happening for a number of customers for different manufacturers - probably based on the fact that with legacy targes there was no easy way to build with baro support but then disable it in configuration, and so users are used to be able to just hook up a baro if I2C pins are available, and it just works.

Probably a good idea to have another read of https://github.com/betaflight/betaflight/blob/master/docs/TargetMaintenance/CreatingAUnifiedTarget.md - we made some clarifications a while ago. I'll try to remember to ping all registered target manufacturers next time this is changed.

kc10kevin, 2020-11-23:
You're most welcome @mikeller.  And sorry for all the issues created with this problem.   I submitted the change request just.  

I completely understand the frustration you encounter sometimes with us.  We do try to keep things cleaned up and up to date, just not as proficient as some and don't keep up like we should.  

I did read up again on the document you mentioned above and did see some of the changes you mention.   I'll make sure I check it periodically to ensure I keep up to speed.

Thanks again and let me know if you need anything further.

maxgame91, 2020-11-24:
I everyone,

Thanks you very much, I modify the unify target for the mamba with the parameter auto for the ligne 87 and it's now detect

So I tried with the BME280 and the dev version and it's detect
I ordered a BMP280 and it's detect with stable version 

Thanks all

ps : I closed this issue

closed 2020-11-24 by maxgame91

etracer65, 2021-02-20:
>Again, thank you for clarifying the issue and I will submit the change to all the Diatone unified Targets to ensure that the Baro and Mag are supported and work out of the box.   At least for the supported Baros/Mags.

@kc10kevin Any updates on when you intend to update the Diatone targets for I2C support - particularly the FURYF4OSD? It continues to be a problem that raises support issues and creates frustration for users.

nwpulver, 2021-03-23:
Hello, I just wanted to mention that the information provided by @etracer65 at the start of this thread solved my problem with the mag not being recognized for both a diatone mamba f722 app FC and a JHMCU BT dual FC.

PR #445 (2021-03-28): Update DIAT_FURYF4OSD to enable I2C baro/mag
Fixes #398 (and many other github/slack/Facebook issues)

Board has I2C pads but target was not configured to allow users to connect baro/mag resulting in many support issues.

Also removed some other inappropriate settings from the target config:

`set rssi_src_frame_errors = ON` - This is a user preference and also based on the type of radio link used and shouldn't be in the target config.

`set dshot_idle_value = 300` - This is a user preference setting based on the motors/ESCs used. This is a quite low value and could cause motor de-syncs for so [... cut]