betaflight/unified-targets #436: Request to change ZEEZ manufacturer information

Guidus93, 2021-02-02:
Hello,
i would like to change ZEEZ manufacturer information:

ID: "ZEEZ" (same as before)
Name: "Zeez RC"
URL: "https://www.zeezrc.com"

I've redirected all from old website to new url so user that will visit old URL will end in new Website.
As before I will be the target maintainer for this manufacturer

Guidus93, 2021-02-22:
@mikeller Sorry to bother, i'm going to release new products and website next days. If is possible to update will be better.
Thanks a lot for your cooperation

mikeller, 2021-02-22:
@Guidus93: The website is still showing 'Sorry, we're doing some work on the site'. :wink: 

But I can update.

Guidus93, 2021-02-22:
Thanks Mike, just finish to upload everything. Wait one or two days to open it as I will open once i will release new products

PR #441 (2021-02-23): Updated information for manufacturer ZEEZ.
Fixes #436.
closed 2021-02-23 by mikeller