betaflight/unified-targets #453: IFlight Succex-E F7 acc/gyro orientation

djtaotao, 2021-03-25:
### Describe the bug
Wrong acc/gyro chip orientation, in "setup" window, roll is pitch and viceversa.

### To Reproduce
Flashed latest firmware, applied custom defaults, calibrate accelerometer and reset z axis to check the proper orientation. I have to set First Gyro to CW 270 to work properly, default is CW 180 which is wrong.

### Expected behavior
Roll to be roll axis and pitch to be pitch axis when move te FC in "setup" window.

### Flight controller configuration
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.8 Feb 15 2021 / 12:10:35 (101738d8e) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F7, version: c7fde51e, date: 2020-08-26T04:30:34Z

# start the command batch
batch start

board_name IFLIGHT_SUCCEX_E_F7
manufacturer_id IFRC

# master
set acc_calibration = -27,19,-12,1
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: FREE
C11: FREE
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 2
    CH2: MOTOR 4
    CH3: MOTOR 1
    CH4: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
- Flight controller:
IFLIGHT SUCCEX-E F7

Linjieqiang, 2021-03-26:
@mikeller Coming now....

Linjieqiang, 2021-03-26:
Since a "unified-targets" config file of ours corresponds to a series of products, not just one hardware board, so in the orientation of the sensor, the user also needs to manually configure the orientation of the sensor according to the provided user manual. 
We do this in order to reduce the maintenance pressure of this warehouse.

mikeller, 2021-03-28:
@Linjieqiang: With Unified Target configurations, there is no extra cost for Betaflight to maintain additional targets, so the limitations that apply to legacy targets apply much less here. This was one of the motivating factors to move to Unified Targets.
This means that it is up to the manufacturer to decide what they want to do in cases like the above:
- add and maintain a single Unified Target configuration for different boards with different sensor orientations, and **provide documentation to users explaining how to configure this**;
- add and maintain multiple different Unified Target configurations for the different gyro orientations, and **tell the user which one to install for each board**.

Linjieqiang, 2021-03-28:
We are more willing to choose the first method. We will indicate the installation direction of the sensor in the user manual and how to set it. In the future, we will unify the sensor installation direction of the board to avoid user confusion.

Linjieqiang, 2021-04-06:
Could the problem be closed?

djtaotao, 2021-04-07:
From my point of view, yes.

closed 2021-04-07 by djtaotao