betaflight/unified-targets #459: TMTR-TMPACERF7 does not have pin assignments for M5/M6 despite having pads for M5/M6

ItsMeAubey, 2021-04-21:
The unified target contains only M1-4.

ItsMeAubey, 2021-04-21:
@tobeornottobe81 @Guidus93 @x4FF3

Guidus93, 2021-04-22:
This is @tobeornottobe81 business, i'm not involved in this board.

ItsMeAubey, 2021-04-22:
My apologies. I just pinged everyone who had submitted patches to TMTR files.

ItsMeAubey, 2021-05-24:
unstale

x4FF3, 2021-07-09:
@ItsMeAubey hi, i created an updated version of this here: https://github.com/x4FF3/unified-targets/blob/tmpacer-motor-fix/configs/default/TMTR-TMPACERF7.config

could you test it?
KR

mikeller, 2021-08-10:
@x4FF3: Do you just want to open a pull request for your modification? If it's only adding more motor outputs then there should not be a risk of it breaking existing setups.

closed 2021-09-20 by github-actions[bot]

AlexZ550Proton, 2022-08-29:
It looks the issue is not fixed. Not sure why. Just in case somebody needs a correct CLI dump I will post it here. It is official dump for resource mapping from TMOTOR support. I tested it and it works great.

board_name TMPACERF7
manufacturer_id TMTR

# name: -

# resources
resource BEEPER 1 C14
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 B07
resource MOTOR 6 B06
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 C15
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 C00
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource COMPASS_CS 1 NONE
resource COMPASS_EXTI 1 NONE
resource SDCARD_CS 1 NONE
resource SDCARD_DETECT 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 B12
resource OSD_CS 1 C13
resource RX_SPI_CS 1 NONE
resource RX_SPI_EXTI 1 NONE
resource RX_SPI_BIND 1 NONE
resource RX_SPI_LED 1 NONE
resource RX_SPI_CC2500_TX_EN 1 NONE
resource RX_SPI_CC2500_LNA_EN 1 NONE
resource RX_SPI_CC2500_ANT_SEL 1 NONE
resource GYRO_EXTI 1 A04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 C04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)

# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2