betaflight/unified-targets #476: Need help converting a BF3.1.5 traget to current version

Guili123, 2021-07-20:
Currently I try to convert the Uvify Draco-Betaflight 3.1.5 Fork firmware to a normal Betaflight Unified target.
I am following the Guide and no big issue so far.
But I noticed some differences in other unified targets compared to the dumps  I have from 3.1.5.
There are no DMA or timer entries, I am wondering if there were differences how this works in Bf3.1.5 and if
this becomes an issue with the new unified targets if these entries are missing.
Is there any way to export DMA und Timer entries with the Uvify (Betaflight 3.1.5) configurator?
Any other way to reverse engineer this?

haslinghuis, 2021-07-20:
A diff only shows the differences from a unified target (+config) it's build on. A dump shows all settings.

jflyper, 2021-07-20:
You should be able to convert basic timer and DMA assignments from target specific `target.c` and `target.h`.

Guili123, 2021-07-20:
I already did a full dump from the Uvify Draco also and the timer+dma information`s are missing in 3.1.5

To clarify this: I have only the dump available. I have no access to the Uvify version from target.c and target.h
The source code for the Uvify fork is not available

haslinghuis, 2021-07-20:
`resource show all`

Guili123, 2021-07-21:
> `resource show all`
Unfortanetly:
# resource show all
Invalid
# 
This is the log output from the configurator:
-- You need to upgrade your firmware to the latest version of Betaflight before you can use the Status tab.
-- Serial port successfully opened with ID: 9
-- Running - OS: Windows, Chrome: 91.0.4472.124, Configurator: 4.0.3
-- Serial port successfully opened with ID: 7
-- Welcome Draco
-- MultiWii API version received - 1.32.0
-- Flight controller info, identifier: BTFL, version: 3.1.5
-- Running firmware released on: Feb 13 2018 09:46:38
-- Board: UVF4, version: 12
-- Unique device ID received - 0x5400313236510a36363331

I can`t access this with the current version of the Betaflight Configurator, because the new version commplains that the MultiWii API need to be upgraded. 

Output in cli:
... dump all
# version
# Betaflight / UVF4_PXBL 3.1.5 Feb 13 2018 / 09:46:38 (7f0a347)
.....
# resources
resource BEEPER 1 A15
resource MOTOR 1 E09
resource MOTOR 2 E14
resource MOTOR 3 E13
resource MOTOR 4 E11
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 C07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 B13
resource SONAR_ECHO 1 B12
resource LED_STRIP 1 A03
...

No entries about DMA and Timer.

jflyper, 2021-07-21:
> To clarify this: I have only the dump available. I have no access to the Uvify version from target.c and target.h
> The source code for the Uvify fork is not available.

In that case, you have to start with `MOTOR` and `LED_STRIP` and guess what target.c looked like for timer and DMA assignments.

Look into appropriate  STM32F4 data sheet, mark pins for motors and led_strip. Look at available TIMx_CHy entries that you can assign to those pins.

For example, PE9, PE11, PE13 and PE14 can be assigned to TIM1, CH1 through CH4 respectively.
Likewise, PA3 has TIM5_CH4.
Then, for DMA, you have to look into corresponding F4 reference manual, DMA request mapping section.
If you can find non-overlapping assignments of DMA streams in that table for TIM1_CH1, TIM1_CH2, TIM1_CH3, TIM1_CH4 and TIM5_CH4, then you are done.
If not, you have to start over again.

Good luck.

jflyper, 2021-07-21:
Btw, this is not a configurator related issue...

Guili123, 2021-07-21:
Thanks, for the hints.
So to summarize, this Uvify configurator version, should be able to list timer and dma.

jflyper, 2021-07-21:
> So to summarize, this Uvify configurator version, should be able to list timer and dma.
I don't think so.
I forgot the exact version, but full timer and dma listing became available much later, probably around 4.0 and later.
You can try something like `timer show` and `dma show` or just `timer` and `dma` to find out on 3.5.x.

mikeller, 2021-07-26:
@Guili123: I suspect that you are looking at an uphill battle here - even if you manage to figure out the timers and DMA, 3.1.5 is so old that most pins for other functions like SPI buses, LED, ADC are not configurable, and won't show in a `dump` (maybe easiest if you provide one here and we can see). So it will require A LOT of guesswork and trial and error to figure out a target for this.

Guili123, 2021-07-26:
@mikeller 
I really appreciate you response here.
At least LED is in the dump...
Not sure if the flight controller is a F745 or F765

Here is a dump:

# dump

# version
# Betaflight / UVF4_PXBL 3.1.5 Feb 13 2018 / 09:46:38 (7f0a347)

# name
name Draco

# resources
resource BEEPER 1 A15
resource MOTOR 1 E09
resource MOTOR 2 E14
resource MOTOR 3 E13
resource MOTOR 4 E11
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 C07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 B13
resource SONAR_ECHO 1 B12
resource LED_STRIP 1 A03

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
[... 420 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 150
set rc_rate_yaw = 150
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 10
set tpa_breakpoint = 165

closed 2021-09-03 by github-actions[bot]