betaflight/unified-targets #493: Detail description for FOXEERF7V2 locked up

DusKing1, 2021-08-12:
Since #482 and #483 were polluted by tons of nonsense, open another issue to provide a detailed problem description here can keep us better at tracking this issue.

Base firmware is 4.2.9, full-chip erases. ESCs are AM32 and Bluejay, both of them support non-bidir dshot and bidir dshot.

#482 can work both in non-bidir dshot and bidir dshot
#483 can work only with bidir dshot, when switch to non-bidir dshot, it will cause the firmware to lock up as I translated [here](https://github.com/betaflight/unified-targets/pull/483#issuecomment-894032351) and the ESC will fill into infinite self-check.

dma show and timer show for #483 when we didn't config the motor protocol (once config the motor protocol and leave bidir dshot as off, then the firmware will lock up):

```
# DMA show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: ADC
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

As I have the hardware sample, I can do any future test.

mikeller, 2021-08-16:
Thanks for the explanation @DusKing1.

Unfortunately this is a completely different outcome to all of the testing that I have been doing, where the proper fix for this (#483) works fine with bidirectional Dshot off (or on). This is only done on a stand-in F722 target, as Foxeer never bothered to provide engineering samples for this board.
Please see this video for a demonstration: https://youtu.be/tZMxmIIYuRk

DusKing1, 2021-08-16:
@mikeller thanks for your reproducing.

However, I found that you are doing this test on 4.3.0 #2494 not the 4.2.9 release. Can you try that unified target?

mikeller, 2021-08-16:
@DusKing1: Thanks for the clarification. I was able to reproduce this, and found that the motor output signal for all ESCs is completely missing when bitbanged Dshot is enabled and bidirectional Dshot is disabled:

[image]

@SteveCEvans: It looks like bitbanged Dshot for non-bidirectional motor output is currently not working - we need to look into fixing this, or else avoid changing targets to use bitbanged Dshot when rearranging resources to free up DMA for SPI.

DieHertz, 2021-08-16:
@mikeller thanks for putting in the effort of hooking up a logic analyzer

mikeller, 2021-08-16:
@DieHertz: No problem - luckily I found a board that used the same motor output pins so it was easy to access them.

SteveCEvans, 2021-08-16:
@mikeller Looking at this using 4.2.9 on a MATEKF722SE. I have motor outputs 4-8 wired up to a logic analyzer in my debug setup, so if I reassign motor 1 using `resource MOTOR 1 B06`, for example that'll use motor output 7. I picked that just because B06 is adjacent to B05, the default pin used for MOTOR 2.

Having set the motor output to a non-zero value in order to see that there's some valid output, with bidirectional DSHOT disabled I see.
[image]

Whereas with it enabled (but without an ESC), I get.
[image]

Much as you'd expect, with the inversion applied.

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# dma adc 1
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
```

What do you believe the issue to be here?

daleckystepan, 2022-01-08:
@DusKing1 @SteveCEvans How about this issue? It is linked to 4.3 Release. Still relevant?

DusKing1, 2022-01-08:
I'm done with FOXEER issue, so IDK

limonspb, 2022-01-10:
@daleckystepan  @SteveCEvans i got foxeer F722V2.
Flashed with 4.3 RC2:

1. when bidir is ON - everything works as expected.
2. When bidir is OFF - escs are doing double first initialization tones, but then spin normally.
DMA and timer CLI are in this file when bidir is OFF:
[foxeer_DMA_timer.txt](https://github.com/betaflight/unified-targets/files/7836988/foxeer_DMA_timer.txt)

One time out of 10 i had motor 1 ESC not initialized and not spinning. The power cycle helped.

by default now dshot_bitbang = AUTO for this target.
If i set it to ON nothing changes in the behavior.
If i change it to OFF - then with bidir=OFF everything works as expected, but can't turn bidir ON.

let me know what kinda test i can also do for you.

limonspb, 2022-01-10:
@SteveCEvans ....
PS: similar behavior for KAKUTEF7MINIV3: (double tones for ESCs when bidir is OFF and sometimes not all escs are being initializaed)
[KAKUTEF7MINIV3_DAM_TIMER.txt](https://github.com/betaflight/unified-targets/files/7837042/KAKUTEF7MINIV3_DAM_TIMER.txt)

Asizon, 2022-01-10:
@limonspb yes we are looking in to this issue, i have reproduced in my non F4 boards and as you said is using bitbanged when not bidir is enabled. Really weird issue but @SteveCEvans doesnt found any solution yet.

limonspb, 2022-01-10:
another possible symptom:
with bidir OFF the ESCs awe weirdly reacting to dshot commands for changing motor direction (from the Configurator dialog):
you send to one ESC to change direction, it does change it but at the same time all three other ESC for a split second increase the throttle, like a tiny full throttle blip. With bidirectional dshot ON everything works well.

limonspb, 2022-01-10:
Flashed 4.3.0 Dec 17 2021 / 23:47:29 (4b44378b4) on Foxerr and there everything works good bidir on/off, no full throttle blips and no double initialization tones.
i guess the issue appeared after this PR is merged?
https://github.com/betaflight/betaflight/pull/11011

SteveCEvans, 2022-01-11:
I've just done a bisect, and I have to agree.
```
f6190e2f1ab3d6d51c4268b6fe707682e7bf547b is the first bad commit
commit f6190e2f1ab3d6d51c4268b6fe707682e7bf547b
Author: Steve Evans <Steve@SCEvans.com>
Date:   Fri Dec 17 11:16:26 2021 +0000

    Use bitbanged DSHOT by default if dshot_bitbang = AUTO unless F4

:040000 040000 20abc0449d5ad4239408b43e3ccd57721aaff87a 62ad7cf6a66a5ac7ce5765d834d35d019d7f24a3 M	src
```

This PR made all non-F4 processors use bitbang DSHOT irrespective of whether bi-drectional mode was enabled. This was to minimise the number of DMA streams used so that SPI_MOSI/SPI_MISO 2/3 would stand a greater chance of being allocated to enable SPI DMA access to FLASH/OSD etc. particularly if burst mode was disabled. On many F7 processors these need to assigned thus:

```
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
```

I don't see this issue on my TMOTORF7 or SPEEDYBEEF7V2, but I do see it on my JHEF7DUAL from which the below are taken.

The prior default with bidirectional DSHOT was
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: ADC
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
And without bidirectional DSHOT.
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: ADC
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

But if I set `set dshot_bitbang=ON` and leave bi-directional DSHOT off we get

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: ADC
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

And we get the double set of beeps.

So actually this behaviour hasn't changed; if `dshot_bitbang=ON` and `dshot_bidir = OFF` then we get the double tones in 4.2.11 too.

Whilst a bit of an oddity, this doesn't warrant including in 4.3

SteveCEvans, 2022-01-11:
The issue reported by @limon in https://github.com/betaflight/unified-targets/issues/493#issuecomment-1008562750 and which I address in my response above is not the same as the original issue I realise having read the entire conversation. Unfortunately the youtube videos are not working for me.

@limon Regarding the motor blip, what's the sequence of commands sent over MSP after `MSP2_SET_MOTOR_OUTPUT_REORDERING`?

closed 2022-02-07 by DusKing1