betaflight/unified-targets #523: HGLRC Zeus 722 hangs after applying config.

obeny, 2021-10-31:
Description:
HGLRC Zeus F722 hangs after power on.

How to reproduce:
Flash latest 4.3 development BF, apply custom defaults when asked.
Board is non-responsive, switching tabs in configurator is not possible.

This problem affects only Zeus F722 FCs only.

For details see:
https://github.com/betaflight/betaflight/issues/11026

HGLRC-D, 2021-11-04:
Do not! This will make the BMP280 on the target not work properly.

jessearmand, 2021-11-05:
Hi @obeny and @HGLRC-D this default configuration is not right, I'm using firmware version 4.2.10, and I had to change the `baro_bustype` into SPI. That's how it was configured before this unified target changes.

Here's what I set to enable the barometer again:
````
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
````

jessearmand, 2021-11-06:
@HGLRC-D As I mentioned elsewhere, I noticed that this config was created without `baro_bustype` being set to any value. 

As can be seen here in [barometer.c](https://github.com/betaflight/betaflight/blob/18af30321dc6ba352f44125fd923f359560a8e5a/src/main/sensors/barometer.c#L155-L273) not having a value there could potentially cause problems. 

But, I haven't tested on the latest 4.3, could you please verify that the BMP280 should work normally with SPI?

closed 2021-12-15 by github-actions[bot]

alienwaredream, 2021-12-23:
> Do not! This will make the BMP280 on the target not work properly.

所以在BF4.3下化骨龙F722飞控到底能不能启用气压计呢？

HGLRC-D, 2021-12-24:
> > 不要！这将使目标上的 BMP280 无法正常工作。
> 
> 所以在BF4.3下化骨龙F722飞控到底能不能启用气压计呢？

可以用的

DusKing1, 2021-12-24:
English please?

alienwaredream, 2021-12-24:
> > > 不要！这将使目标上的 BMP280 无法正常工作。
> > 
> > 
> > 所以在BF4.3下化骨龙F722飞控到底能不能启用气压计呢？
> 
> 可以用的

我试了在4.2下用

HGLRC-D, 2021-12-24:
I tested 4.3 here and there is no problem. Have you tried to re-burn the firmware?

> English please?

Sorry, i will pay attention.

HGLRC-D, 2021-12-24:
> > > > 不要！这将使目标上的 BMP280 无法正常工作。
> > > 
> > > 
> > > 所以在BF4.3下化骨龙F722飞控到底能不能启用气压计呢？
> > 
> > 
> > 可以用的
> 
> 我试了在4.2下用 set baro_bustype = SPI set baro_spi_device = 1 可以正常启用气压计
> 
> 但是在4.3下，只要一输入set baro_bustype = SPI 保存之后飞控重启就死机了，是不是和4.3对SPI总线定义有冲突？

Have you tried boot?
Confirm "set baro_bustype = SPI
set baro_spi_device = 1" Complete input?
I tested 4.30 , and baro can be used normally.
If the attempt fails, I will ask my colleague to contact you.

HGLRC-D, 2021-12-28:
> > > > 不要！这将使目标上的 BMP280 无法正常工作。
> > > 
> > > 
> > > 所以在BF4.3下化骨龙F722飞控到底能不能启用气压计呢？
> > 
> > 
> > 可以用的
> 
> 我试了在4.2下用 set baro_bustype = SPI set baro_spi_device = 1 可以正常启用气压计
> 
> 但是在4.3下，只要一输入set baro_bustype = SPI 保存之后飞控重启就死机了，是不是和4.3对SPI总线定义有冲突？

You can try to enter:
“
set baro_i2c_device = 0
set baro_bustype =SPI
set baro_spi_device = 1
save
”