betaflight/unified-targets #526: New Manufacturer ID for Axisflying

x4FF3, 2021-11-06:
Hi,

i would like to request a new manufacturer id for: Axisflying
Website: axisflying.com

KR
David

DusKing1, 2021-11-10:
Is this a Chinese company or established by certain Japanese fascist remnant bastards?

I was just doing some random search and found this: https://youtu.be/x8rqGuSpNwI?t=3
[image]

This is really a terrifying thing TBH, even 76 years after the surrender of the Japanese fascists, there are still people who follow them as their example. And the "Axisflying" offer support to him?! 

GitHub is not a place to discuss politics, but I think we should not tolerate any fascist remnants. No offence to any decent Japanese programmer to be clear.

Jesus!!

x4FF3, 2021-11-10:
Hi, 

as i do not directly represent Axisflying, nor i will speak for them. So please get in contact with them about this. My personal comment on this. This is a random youtube video of a FPV pilot, using a certain brand. I dont think any of the used brands where involved in making any decision regarding this video.

KR
David

Axisfiying, 2021-11-10:
> Is this a Chinese company or established by certain Japanese fascist remnant bastards?
> 
> I was just doing some random search and found this: https://youtu.be/x8rqGuSpNwI?t=3 [image] If you don't know what this flag is please see https://en.wikipedia.org/wiki/Rising_Sun_Flag
> 
> This is really a terrifying thing TBH, even 76 years after the surrender of the Japanese fascists, there are still people who follow them as their example. And the "Axisflying" offer support to him?!
> 
> GitHub is not a place to discuss politics, but I think we should not tolerate any fascist remnants. No offence to any decent Japanese programmer to be clear.
> 
> Jesus!!

We have dealt with this problem. Thank you for your attention

DusKing1, 2021-11-10:
@Axisfiying what the fck are you talking about? Do you understand English?

luttesudo, 2021-11-11:
@DusKing1  Last time it was iFlight, this time it was Axisflying. Would you like to turn Github into a public opinion forum?

davecme, 2021-11-11:
@luttesudo from what I saw @DusKing1 is a collaborative contributor to this project and @luttesudo you are not. What makes you said those?

luttesudo, 2021-11-11:
@DusKing1 Because you should reply directly, you should not reply with the account you just registered. By the way, as an administrator, you should pay more attention to your words and actions in the community instead of spreading public opinion everywhere.

DusKing1, 2021-11-11:
It seems like both @luttesudo and @davecme are fake registers
[image]

luttesudo, 2021-11-11:
Oh, is it so? You have a lot of code, and you shouldn’t publish things that don’t belong here on Github. As an administrator, you should manage these things that don’t belong to github, instead of using your rights to make a splash here.

ctzsnooze, 2021-11-23:
The views expressed by Dusking reflect his personal perspective on what he clearly finds a sensitive topic.  
His language and manner of speech reflect the strength of his feelings but are inappropriate in this context and do not reflect the standards of behaviour that we in Betaflight aspire to.
I am Australian and am well aware of the circumstances of the war in the Pacific, having lost my great-uncle in that war.
The owner of this company may reflect on the fact that the Japanese, Italian and German forces in the second world war were known as the 'Axis' forces.  
The coincidence of that name and that flag in the video are most unfortunate.
Having said that, I would be grateful if further discussion of the use of this symbol and the company name could take place privately, not here.

closed 2021-12-29 by haslinghuis