betaflight/unified-targets #595: Add manufacturer id for DAKEFPV

ThanYangFPV, 2022-04-30:
Hi.
I want to apply for a manufacturer id for dakefpv. Here is our information.

Manufacturer name : DAKEFPV
Mail : sals@dakefpv.com
Website : [https://dakefpv.com/](https://dakefpv.com/)
ID : DAKE

If the application gets approved, I want to be his maintainer.
Thanks

Asizon, 2022-04-30:
@ThanYangFPV  where is your pull request for hardware to maintain? Fast look in the website and seems that the Fcs comes from betafpv.

ThanYangFPV, 2022-04-30:
> @ThanYangFPV where is your pull request for hardware to maintain? Fast look in the website and seems that the Fcs comes from betafpv.

Yes.We are proxying their product, this is temporary, once we have an ID, we will pull request belonging to our target, and then start making our product.

Dakefpv, 2022-05-29:
@Asizon Hi，Asizon，Nice to meet you! I am Jason from DAKEFPV，I authorize thanyangfpv to maintain our flight control hardware!

closed 2022-06-05 by Asizon