betaflight/unified-targets #621: Add manufacturer id for GEELANG

Linjieqiang, 2022-06-10:
Hi all.
I want to apply for a manufacturer id for GEELANG. Here is our information.

Manufacturer name : GEELANG
Website : www.geelang.com
ID : GEPC

If the application gets approved, I want to be his maintainer.
Thanks

haslinghuis, 2022-06-10:
The store really needs some updates as it's main purpose seems to be selling PC stuff instead of RC.

Linjieqiang, 2022-06-11:
[image]

KarateBrot, 2022-06-11:
Looks more like a spam site to be honest. If you look at Flight controllers there are no flight controllers at all.
[image]