betaflight/unified-targets #641: Default Gyro Alignment is not correct on TMOTOR F411 Lite unified target.

liu-root, 2022-07-07:
[image]

## Gyro not working on default configuration for the TMOTOR F411 Lite FC. 
1. Gyro alignment is incorrect, **I had to change the YAW to alignment to 270 degrees.** 
1.1 FC is mounted correctly with the arrow pointing to the front of the quad. 
1.2 I also noticed that the default accelerometer position had the 3D model upside down (before the calibration). 
3. Default installation of the 4.3.0 shows the issue right after installation. 
4. Hardware used: T-Motor Velox F411 Lite 30x30 Flight Controller - https://store.tmotor.com/goods.php?id=1229
3.1 Betaflight 4.3.0
3.2 Unified Config: https://github.com/betaflight/unified-targets/blob/master/configs/default/TMTR-TMOTORF411.config
3.3: Diff:
```
# beacon
beacon RX_LOST

# map
map TAER1234

# aux
aux 0 0 4 900 1300 0 0
aux 1 1 0 1700 2100 0 0
aux 2 2 0 1275 1675 0 0
aux 3 13 1 1700 2100 0 0
aux 4 35 1 1300 1700 0 0

# master
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 2
set dyn_notch_q = 400
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 300
set gyro_lpf1_dyn_max_hz = 600
set gyro_lpf1_dyn_expo = 8
set acc_calibration = -16,-8,51,1
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set rssi_channel = 16
set rc_smoothing_auto_factor = 120
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set align_board_yaw = 270
set ibata_scale = 80
set small_angle = 180
set simplified_gyro_filter = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_vbat_pos = 2507
set osd_rssi_pos = 2068
set osd_link_quality_pos = 2139
set osd_rssi_dbm_pos = 121
set osd_tim_2_pos = 2081
set osd_flymode_pos = 2061
set osd_throttle_pos = 122
set osd_vtx_channel_pos = 128
set osd_current_pos = 2500
set osd_mah_drawn_pos = 2516
set osd_craft_name_pos = 2052
set osd_gps_speed_pos = 0
set osd_gps_lon_pos = 481
set osd_gps_lat_pos = 497
set osd_gps_sats_pos = 26
set osd_flight_dist_pos = 2112
set osd_warnings_pos = 14635
set osd_avg_cell_voltage_pos = 2512
set osd_disarmed_pos = 2155
set osd_flip_arrow_pos = 2318
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set gyro_1_sensor_align = DEFAULT
set rpm_filter_harmonics = 1
set name = Mark 4 HD

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_static_hz = 80
set dterm_lpf2_static_hz = 140
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set simplified_dterm_filter = OFF

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

haslinghuis, 2022-07-13:
@Guidus93 if this board is to be targeted for Betaflight could you please spin up an unified target config.

Guidus93, 2022-07-14:
@haslinghuis yeah will push a new unified target.

Guidus93, 2022-07-15:
Fixed with [PR #653](https://github.com/betaflight/unified-targets/pull/653)

closed 2022-07-15 by haslinghuis

liu-root, 2022-07-15:
Thank you gentlemen! You guys rock!

Guidus93, 2022-07-15:
> Thank you gentlemen! You guys rock!

You’re welcome sorry for the inconvenience