betaflight/unified-targets #773: DarwinFPV SX1280 board lacks OSD

howels, 2023-01-02:
@DMAXYANG the DarwinFPV AIO SPI based eLRS boards seem to be using the target https://github.com/betaflight/unified-targets/blob/master/configs/default/DAFP-DARWINF4SX1280HD.config which lacks OSD support.
Understand that there may exist a board without OSD chip, but DarwinFPV sell SX1280 boards with OSD and these are lacking support currently.
Should we add `#define USE_MAX7456` to DARWINF4SX1280HD or create a new target for OSD-enabled versions of the board?

haslinghuis, 2023-01-02:
Adding OSD driver will fix it.

howels, 2023-01-02:
> Adding OSD driver will fix it.

Yes, but target is specifically named after the HD variant so the target name could confuse pilots

closed 2023-01-02 by blckmn