betaflight/unified-targets #798: Lost Gyro on MAMBAF722_I2C Target in 4.4.0 RC

peewiz, 2023-01-08:
When upgrading MAMBAF722_I2C from 4.3.2 to 4.4.0 RC(1,2 or 3)  gyro is not available anymore on MAMBAF722_I2C target

By comparison, I can see that this define was added in #771 
#define USE_BARO_BMP280
but it is in accordance with barometer of this board.

Here is the result of status command, and the version command showing board name

# Status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=31degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4034, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
**GYRO=MPU6500**, ACC=MPU6500, BARO=BMP280
OSD: MAX7456
System Uptime: 9 seconds, Current Time: 2023-01-08T16:32:40.651+00:00
CPU:31%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 8
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.2 Nov 28 2022 / 07:30:19 (60c9521) MSP API: 1.44
# config: YES
# board: manufacturer_id: DIAT, board_name: **MAMBAF722_I2C**

Status shows that gyro of this board (Mamba DJI F722 MK2) is MPU6500 and not MPU6000, maybe the reason ? I have no other clue.
If I can be of any help !

Thank you !

haslinghuis, 2023-01-08:
Please add `GYRO_SPI_MPU6500` and `ACC_SPI_MPU6500` in custom defines and report status

peewiz, 2023-01-08:
Hi Haslinghuis
You are the best ! 
I confirm that adding GYRO_SPI_MPU6500 and ACC_SPI_MPU6500 in custom defines solves the gyro availability !

haslinghuis, 2023-01-08:
Thanks, will add them to the target.