betaflight/unified-targets #832: Problem on DarwinFPV with BF 4.4

Denethor85, 2023-01-22:


FulvioSpelta, 2023-03-31:
Hi all, I've just flashed the 4.4 using the configurator and the gyroscope is now not detected.
How's the state of this issue ? Is there a workaround ?

Thanks a lot.

haslinghuis, 2023-03-31:
> here is the PR to fix this issue https://github.com/betaflight/unified-targets/pull/918

haslinghuis, 2023-03-31:
Please use DARWINF411 target instead.

You might have to change in CLI:

    set gyro_1_sensor_align = CW180
    set gyro_1_align_pitch = 0
    set gyro_1_align_yaw = 0
    save

FulvioSpelta, 2023-03-31:
OK, I've already flashed the controller using the DARWINF411 target that is automatically detected by the configurator.
Regardings the defauls I've deselected GPS that isn't present and then flashed.

Later (sorry I've not here with me the drone) I'll apply the sets you just suggested and i'll provide e feedback.

closed 2023-05-09 by github-actions[bot]