betaflight/unified-targets #877: BF 4.4 Blackbox Flash Chip Not Recognized - HAKRC F7230D

Warl0rd0ne, 2023-02-05:
[BTFL_cli_20230205_023831.txt](https://github.com/betaflight/unified-targets/files/10610523/BTFL_cli_20230205_023831.txt)
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

Warl0rd0ne, 2023-02-05:
version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 03:03:40 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: MTKS, board_name: MATEKF722SE

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=45degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3791, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma shared
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456 (30 x 13)
BUILD KEY: af774d45994ff10105100b4051902a18 (4.4.0)
System Uptime: 46 seconds, Current Time: 2023-02-05T08:32:55.698+00:00
CPU:25%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 31 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 2
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 03:03:40 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A01
resource PWM 3 A00
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
[... 271 lines cut ...]
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW90
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 900
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

# 
*** Support data submitted *** 
Id: "5af98a9e-931c-42c8-9057-d61e1e6cd05e"

haslinghuis, 2023-02-05:
Issue seems to be here: `SD card: Startup failed`

Matek F722SE uses a real SD Card. HAK7230D has a 16MB blackbox chip. Please try:

```
set blackbox_device = SPIFLASH
save
```

Warl0rd0ne, 2023-02-05:
Completed Above, but change did not stick. Reverted to blackbox_device = NONE.
in 4.3 the flash chip did not work out of the box either, but was able to make it work by reassigned resource pins and then changing the above setting.
I tried reassigning the pin, but this is the output I got:

# get blackbox_device
blackbox_device = NONE
Allowed values: NONE, SPIFLASH, SDCARD, SERIAL
Default value: SDCARD

# resource SDCARD_CS 1 NONE
Resource is freed

# resource FLASH_CS 1 D02
###ERROR IN resource: INVALID RESOURCE NAME: 'FLASH_CS'###

# 

Think there is a missing define in the unified target config. Tried inputting a custom define at build time (Cloud Build), but it would not add it. Maybe the custom define section can only add software features and not hardware drivers?

I tried using a different unified target "HARC-HAKRCF722D". Looks almost identical to the MTKS-MATEKF722SE config except for it has some different hardware defines.
The Flash Chip worked with this config, but the Baro did not. Missing define: USE_BARO_BMP280
again tried putting this define into the custom defines box, but it would not add it to the cloud build.

Warl0rd0ne, 2023-02-05:
Here is the Core build of the HARC-HAKRCF722D config support info:

version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 03:02:23 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: HARC, board_name: HAKRCF722D

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=47degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3791, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma shared
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456 (30 x 13)
BUILD KEY: c7f4476a99098f9d07454d66314506e7 (4.4.0)
System Uptime: 69 seconds, Current Time: 2023-02-05T18:19:30.608+00:00
CPU:25%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 32 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 9
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS ANGLE CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 03:02:23 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HAKRCF722D
manufacturer_id HARC

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A01
resource PWM 3 A00
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
[... 279 lines cut ...]
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW90
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 900
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

# 
*** Support data submitted *** 
Id: "df0541aa-ef2f-4486-9cc8-ec78fb52a467"

Warl0rd0ne, 2023-02-05:
The Baro Worked with the core build along with the flash chip working. This is on the HAKRCF722D target.

The only other issue I see right now is that the gyro board alignment is not correct - HAKRCF722D config has it set to CW180 when MATEKF722SE confgi has it set to CW180Flip. Which CW180Flip is correct for HAKRC F7230D Flight Controller.

haslinghuis, 2023-02-05:
Thanks created a new target based on this information.

PR #879 (2023-02-08): Add HARC-HAKRCF7230D
Fixes: #877
closed 2023-02-08 by haslinghuis