betaflight/unified-targets #921: HGLRC F722 mini and BF 4.5-zulu are not friends.

LoakAudio, 2023-02-25:
Hi all,

I'd like to test PR #12359, but my HGLRCF722 mini isn't OK with 4.5-zulu.
Link to the product page : https://www.hglrc.com/collections/fc/products/hglrc-zeus-f722-3-6s-f722-flight-controller
It seems to be a new product as the electronic components on the PCB are not in the same place as on my previous card of the same model.

So what happens ?
- Compiling the hex for HGLRCF722E (the one that was working for my previous card) ends to no ACC, no GYRO and no BARO, even with ACC_BMI270 GYRO_BMI270 BARO_DPS310 in the custom defines (seems to be the good  ones if I trust the "core only" build).
  Link to the log of the build https://build.betaflight.com/api/builds/0ec55e6dd18e4438b81487cb39ff7a16/log

- Compiling the hex for HGLRC722 (non-E) ends with no BARO.

- Compiling the hex for HGLRC722 (non-E) and BARO_SPI_DPS310 or BARO_DPS310 in custom defines ends with a failed
  Link to the log of the failed build : https://build.betaflight.com/api/builds/a73b186224fca2c664f29c56948f09db/log

After each flash, default configuration for the card has been applied.

Status of a "core only" build for HGLRCF722E :
(BARO is missing, ACC OK, GYRO OK, FLASH OK)

```
# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=36degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3792, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma shared
GYRO=BMI270, ACC=BMI270
OSD: NONE (68 x 12)
BUILD KEY: b5ee946dca4d220dc7b444af592581b2 (4.5.0-zulu)
System Uptime: 19 seconds, Current Time: 2023-02-25T18:20:05.009+00:00
CPU:10%, cycle time: 308, GYRO rate: 3246, RX rate: 15, System rate: 9
Voltage: 28 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 7
SD card: Not configured
FLASH: JEDEC ID=0x00ef4017 8M
Arming disable flags: RXLOSS ANGLE CLI MSP MOTOR_PROTO
```

Status for a "core only" build for HGLRCF722 (non-E) :
(seems to be the right one for this version of the card, nothing missing)

```
# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=38degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3792, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: NONE (68 x 12)
BUILD KEY: a05d6d20d47451a879979707f9deb887 (4.5.0-zulu)
System Uptime: 173 seconds, Current Time: 2023-02-25T18:28:31.709+00:00
CPU:12%, cycle time: 308, GYRO rate: 3246, RX rate: 15, System rate: 10
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 9
SD card: Not configured
FLASH: JEDEC ID=0x00ef4017 8M
Arming disable flags: RXLOSS ANGLE CLI MSP

```

With Betaflight 4.4, everything is OK with HGLRCF422 target.

Let me know if you need more information.

haslinghuis, 2023-02-25:
Duplicate of #12402

LoakAudio, 2023-02-25:
@haslinghuis OK, sorry, Do I delete it ? or close it ?

haslinghuis, 2023-02-25:
Let's keep it as reminder - noticed BARO was not working too.

closed 2023-04-05 by github-actions[bot]