betaflight/unified-targets #958: No barometer BMP280 for FLYWOOF411 via BF4.4 Cloud Build Config

mxfre, 2023-03-21:
1. Barometer BMP280 cannot be enabled on NOXE V1 (deluxe version) Flight Controller, target FLYWOOF411, via BF4.4 Cloud Build Config.
Previously in BF4.1 it can be enabled in config and cli.

https://www.amazon.ae/Upgraded-20x20mm-Controller-Barometer-Blackbox/dp/B087X7KRZR

2. To Reproduce

Update NOXE V1 flight controller to BF4.4 (Firmware Target: FLYWOOF411) via BF Configurator 10.9.
Enable "Barometer" in BF Configurator 10.9.

Expected behavior

On Board Blackbox shall be available in the BF Configurator 10.9 - Blackbox tab option.

Baro works in "CORE only", but I need GPS.

I have tried Cloud Build Mode adding "BARO_BMP280" in the custom defines field, but still Baro are not available/detected.
I think the problem in the Cloud Build is due to the Unified Target for FLYWOOF411 was not updated with:

#Define USE_BARO_BMP280 

https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF411.config

haslinghuis, 2023-03-22:
To use custom defines please enable show release candidates and then enable expert mode.
If `BARO_BMP280` still does not work also try `BARO BARO_BMP280`

mxfre, 2023-03-22:
It's the same, BARO is not recognized.
As I said, I think the problem in the Cloud Build is due to the Unified Target for FLYWOOF411 was not updated with:

#Define USE_BARO_BMP280

https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF411.config

haslinghuis, 2023-03-22:
The original FLYWOOF411 has no baro. Our policy is not to add drivers not used on the original target.

[image]

Please share the support ID after flashing core build.

mxfre, 2023-03-27:
FLYWOOF411 is not the same as FLYWOOF411 v2.1.
On my NOXE V1 (deluxe version), I flashed Betaflight 4.2.11 firmware for target FLYWOOF411.
BARO is available.
[image]

[image]

[dump_Baro_4.2.11.txt](https://github.com/betaflight/unified-targets/files/11082145/dump_Baro_4.2.11.txt)

mxfre, 2023-04-21:
I just noticed you added "#define USE_BARO_BMP280" to the Unified Target for FLYWOOF411.
Baro works now by adding "BARO BARO_BMP280" to custom defines.
Thank you.

haslinghuis, 2023-04-21:
Fixed in #977

closed 2023-04-21 by haslinghuis