betaflight/unified-targets #997: 'No Dataflash chip found' on Neutronrc NEUTRONRCF411SX1280 AIO

jumpstartfpv, 2023-04-30:
1. Dataflash chip is present on Betaflight 4.3.2 and the output of the 'status' command in cli is as under

MCU F411 Clock=108MHz (PLLP-HSI), Vref=3.29V, Core temp=46degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3994, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM20689, ACC=ICM20689
OSD: NONE
System Uptime: 9 seconds, Current Time: 2023-04-30T10:03:32.455+00:00
CPU:32%, cycle time: 122, GYRO rate: 8196, RX rate: 15, System rate: 9
Voltage: 74 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS ANGLE CALIB CLI MSP

2. Dataflash chip is not detected on bf 4.4.1

3.  When flashed with the 'core' build of the 4.4.1 data flash is detected and 'status' is as under

MCU F411 Clock=108MHz (PLLP-HSI), Vref=3.29V, Core temp=49degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3752, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM20689, ACC=ICM20689
OSD: NONE (133 x 213)
BUILD KEY: a59458b45ac5159d7d23d012e4490d26 (4.4.1)
System Uptime: 13 seconds, Current Time: 2023-04-30T09:55:42.842+00:00
CPU:31%, cycle time: 122, GYRO rate: 8196, RX rate: 15, System rate: 9
Voltage: 82 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS ANGLE CLI MSP

Support Data Uploaded with ID
62bb2c8c-4ccd-4bb7-ac77-8762f12de2bd

4. No access to the ESCs on the AIO when 4.4.1 is flashed. To detect esc in esc-configurator I had to flash back to 4.3.2

nerdCopter, 2023-04-30:
temporary fix for flash: Enable expert mode on the flashing tab and add `FLASH_W25Q128FV` to custom defines.
if that fails, try `FLASH FLASH_W25Q128FV`, but should not be needed.

[image]

please reply back if it works, devs will add that flash-chip to the defines.

item `#4`. please select `development` and add `4.4-maintenance` as the screenshot shows to see if this solves the ESC.  please report status.

jumpstartfpv, 2023-05-02:
Added the custom define **FLASH_W25Q128FV**

Flashed using these settings on the latest release configurator. No change. No Dataflash chip detected.

[image]

nerdCopter, 2023-05-02:
there is this define as well -- unsure if such is the solution.
[image]

jumpstartfpv, 2023-05-02:
> there is this define as well -- unsure if such is the solution. [image]

Unable to find this option.

I installed the latest nightly configurator and flashed 4.5 Zulu with the flash config and latest master but still do not have the ESC commuication.

nerdCopter, 2023-05-02:
jsut have to click around
[image]

jumpstartfpv, 2023-05-03:
Flashed both 4.4 and 4.5zulu with custom option of **ESC Serial** but no communication with esc using latest web version of esc configurator.

Again confirmed that it works on 4.3 but not on 4.4. using latest nightly build of configurator.

haslinghuis, 2023-05-03:
ESC Serial (SK) Inc 4way is for SimonK firmware and 4 way protocol used for a programmer like an Arduino so this would not be needed at all. (Even APD does not use it anymore).

Need to close all serial connections before using esc-configurator.com - connect to com port - and attach lipo to read / write settings. Please also try with 4.5.0

PR #999 (2023-05-11): add W25Q128FV to NEUTRONRCF411SX1280
* add W25Q128FV to NEUTRONRCF411SX1280
* https://github.com/betaflight/unified-targets/issues/997#issuecomment-1531982367
* https://discord.com/channels/868013470023548938/924169080045445120/1103035100515549314

closes #997
closed 2023-05-11 by haslinghuis