betaflight/unified-targets #1008: Joshua Bardwell F4 AIO V2 (CLRACINGF4) Dataflash Not Recognized

HCoop22, 2023-05-15:
1. Describe your problem;
When updating to Betaflight 4.4.1 the flight controller onboard dataflash is not recognized and I am unable to blackbox log.
2. Include ways to reproduce the problem;
Flash Betaflight 4.3.2 (works correctly) and then flash Betaflight 4.4.1
3. Provide as much information as possible about your hardware and software, including:
Joshua Bardwell F4 AIO V2
Betaflight V4.4.1
Betaflight Configurator V10.9.0
*** Support data submitted *** 
Id: "d45fa721-2d22-48fd-9462-74c2aebb16d8"

haslinghuis, 2023-05-16:
We are awaiting an answer here: https://github.com/betaflight/unified-targets/pull/925

HCoop22, 2023-05-16:
Mark,
Thanks for the response, I’ll keep an eye on that thread.
Also as an FYI I have tried the putting the following in the target config box while flashing but none have worked.

FLASH_W25P16
FLASH_W25M512
FLASH_W25Q128FV
FLASH_W25N01G
FLASH_W25M02G

I’m assuming I did the process correct (select BF4.4.1, put one of the FLASH numbers in the target config, flash flight controller, apply custom defaults, still no data flash, try new flash number, reflash fc, repeat)

Thanks,
Haydon

nerdCopter, 2023-05-16:
@HCoop22 , Configurator 10.9 has a bug that ignores custom defines. Mitigation is to enable export-mode, or use a nightly build of Configurator.
[image]

HCoop22, 2023-05-17:
@nerdCopter I tried the list of custom defines in my last post with expert mode turned on and none of them worked, dataflash still not recognized. Any other custom defines I should try?

PR #1047 (2023-06-19): Add CLRACINGF4V2
Fixes #1008
closed 2023-06-19 by haslinghuis