betaflight/unified-targets #1020: Aocoda F7 Mini Flight Controller Missing Pad

StatErik, 2023-05-27:
The configuration file for the Aocoda F7 Mini Flight Controller seems to be confusing and lacks a pad in the setup. Following the provided wiring instructions, I have measured the following connections:

- S5 - B06
- S6 - B03
- S7 - A15
- S8 - B07

**Problem 1: Missing Pad Assignment for S8**
The configuration file completely overlooks the assignment of Pad S8 to Pin B07. It would be helpful if motor 8 could be assigned to this pad in the configuration.

**Problem 2: Inconsistent Motor Assignments**
Based on the wiring diagram, I assumed that motor 5 should be connected to S5, motor 6 to S6, and so on. However, the current configuration does not align with this assumption, causing confusion and hindering the correct setup process.

**Expected Behavior:**
To address the issues mentioned above, I suggest the following:

- Include the assignment of Pad S8 to Pin B07 in the configuration file to enable the use of Motor 8.
- Ensure that the motor assignments in the configuration file match the pad numbers in the corresponding wiring instructions for a more intuitive setup process.

**Additional Information:**

- Aocoda F7 Mini Flight Controller model version: 1.0
- Bought at: Aliexpress - Readytosky
- My FC bricked after I flashed the cloud built Betaflight to it. It doesn't even start in DFU-Mode with the Boot Button pressed. I haven't found the issue yet. Be warned!

[image]

Please let me know if you need any further information or clarification regarding this issue.

nerdCopter, 2023-05-27:
@aocodarc

aocodarc, 2023-06-01:
There is such a mistake indeed. I have prepared to submit pr by adding s8 motors in the correct sequence of motors