betaflight/unified-targets #1034: BetaFPV F4 1S 12A AIO V2.2

hi7mansyah, 2023-06-05:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem; When updating to Betaflight 4.4.2 the flight controller, in tab sensor gyroscope line is flat, but accelerometer is  OK
3. Include ways to reproduce the problem; Flash Betaflight 4.3.1 (works correctly) and then flash Betaflight 4.4.2
4. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using; Betafpv F4 1S 12A AIO V2.2
- the version of all the software used; Betaflight Configurator V10.9.0
- how things are connected / wired up.

Id: "3c11f428-403f-453a-bf71-9fe97da06c22"

haslinghuis, 2023-06-05:
GYRO=BMI270, ACC=BMI270 are both present in status command. So it looks like a firmware issue as I only have the ICM20689 and ICM42688P BETAFPVF4SX1280 for testing. Tested some other board (TUNERC405) with BMI270 and found no problem.

So it looks like this (as this test board was DOA):
[image]

hi7mansyah, 2023-06-05:
I've tried flashing again, the result is still the same. Btw thanks for the information. Maybe the Gyro has a problem
[image]

hi7mansyah, 2023-06-05:
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.28V, Core temp=44degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3606, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: e694c9fef11de5387f5e11f6007e40d8 (4.4.2)
System Uptime: 389 seconds, Current Time: 2023-06-05T22:39:44.822+00:00
CPU:22%, cycle time: 312, GYRO rate: 3205, RX rate: 15, System rate: 9
Voltage: 2 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CALIB CLI MSP

haslinghuis, 2023-06-05:
> CALIB 

Did you calibrate?

Also try 4.5.0 preview

hi7mansyah, 2023-06-05:
I have calibrated, the result remains the same...,
I will try 4.5.0, thanks

hi7mansyah, 2023-06-09:
it's working now...
[image]

closed 2023-06-09 by hi7mansyah

haslinghuis, 2023-06-09:
@liyouluo `dshot_bitbang` is disabled in the target, should we update the target?