betaflight/unified-targets #1054: Darwin FPV No Gyro/ACC for MATEKF411 v4.4.2 - Should be MPU6500 - Id: "5fd47ee8-779b-4ea4-844a-ab77ab5e97a9"

KennethEMarks, 2023-06-24:
Id: "5fd47ee8-779b-4ea4-844a-ab77ab5e97a9"

When flashing online (non-core) Betaflight 4.4.2 target MATEKF411 to Darwin FPV MATEXF411 FC, No Gyro/Acc error occurs. Core build shows in fact that Gyro/Acc MPU6500 is present on board. See attached Support Data dump from Betaflight Configurator v10.9.0:

2023-06-23 @12:26:28 -- OS: macOS

2023-06-23 @12:26:28 -- Configurator: 10.9.0 (697943b)

2023-06-23 @12:26:28 -- Loaded release information for configurator from GitHub.

2023-06-23 @12:26:39 -- Serial port successfully opened with ID: 1

2023-06-23 @12:26:39 -- MultiWii API version: 1.43.0

2023-06-23 @12:26:39 -- Flight controller info, identifier: BTFL, version: 4.2.5

2023-06-23 @12:26:39 -- Running firmware released on: Nov 22 2020 18:38:45

2023-06-23 @12:26:39 -- Board: MTKS/MATEKF411(STM32F411), version: 0

2023-06-23 @12:26:39 -- Unique device ID: 0x1d003e3132510633353639

2023-06-23 @12:26:39 -- Craft name: Baby Ape

2023-06-23 @12:26:39 -- Arming Disabled

2023-06-23 @12:27:07 -- Flight controller set Rateprofile: 2

2023-06-23 @12:27:08 -- Flight controller set Rateprofile: 3

2023-06-23 @12:27:10 -- Flight controller set Rateprofile: 1

2023-06-23 @12:27:28 -- Flight controller set Rateprofile: 2

2023-06-23 @12:27:30 -- Flight controller set Rateprofile: 1

2023-06-23 @12:27:53 -- Flight controller set Rateprofile: 2

2023-06-23 @12:27:54 -- Flight controller set Rateprofile: 1

2023-06-23 @12:28:40 -- Flight controller set Rateprofile: 2

2023-06-23 @12:29:06 -- Flight controller set Rateprofile: 1

2023-06-23 @12:29:07 -- Flight controller set Rateprofile: 2

2023-06-23 @12:29:20 -- Flight controller set Rateprofile: 3

2023-06-23 @12:29:43 -- Flight controller set Rateprofile: 2

2023-06-23 @12:29:45 -- Flight controller set Rateprofile: 3

2023-06-23 @12:29:47 -- Flight controller set Rateprofile: 1

2023-06-23 @12:29:57 -- Flight controller set Rateprofile: 2

2023-06-23 @12:29:59 -- Flight controller set Rateprofile: 3

2023-06-23 @12:30:01 -- Flight controller set Rateprofile: 2

2023-06-23 @12:30:03 -- Flight controller set Rateprofile: 1

2023-06-23 @12:30:23 -- Flight controller set Rateprofile: 2

2023-06-23 @12:31:02 -- EEPROM saved

2023-06-23 @12:32:58 -- EEPROM saved

2023-06-23 @12:33:10 -- Flight controller set Rateprofile: 1

[... 1306 lines cut ...]
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 1
set scheduler_relax_osd = 1
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "5fd47ee8-779b-4ea4-844a-ab77ab5e97a9"

# copy ID and provide to the betaflight team.

haslinghuis, 2023-06-24:
Did you try DARWINF411 ?

KennethEMarks, 2023-06-24:
Awe snap!

I did not, but just found it now. I was going from Darwin’s instructions and some online help. I figured this was something that probably was already solved by the team, but I figured I would file an issue just in case.

Thank you so much for getting back to me so fast, and for all you are doing for the community! If it is easier, I can post a query in the discord in the future before filing an issue report.

Best regards,

- Ken Marks

closed 2023-08-03 by github-actions[bot]