betaflight/unified-targets #1090: Lost of Gyro,Acc and Barometer when upgrading to bf 4.4.2

Maxluli, 2023-08-09:
Hello
I just received my new FPV drone, which is the Flywoo Explorer lr4.
It has the Flywoo F405 Stack
When plugged into betaflight everything worked on betaflight 4.3.2, but I wanted to install the newest version (4.4.2 t the time) of Betaflight, so I could use my M10 GPS at its fullest potential, but after flashing the new firmware... No more gyro, accelerometer or barometer.

I watched this video to fix this issue : 
https://www.youtube.com/watch?v=ksxONEal7yo&ab_channel=BaconNinjaFPV

I created a dump file in the CLI with this code : 
Id: "30bef207-fe91-45c0-87aa-3f01d0b14de0" 

But here is for example the begining of my status : 

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=51degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP

I added 

"GYRO_SPI_ICM42688P ACC_SPI_ICM42688P BARO_DPS310"

To the custom defines in the firmware flasher... But after flashing nothing showed up and after loading my old backup gyro et acc showed up but not the barometer.

I would like to know if i did something wrong and if it ccould be possible to add the correct defines so the next person to have its lr4 explorer does not have the same issue. Thanks !

nerdCopter, 2023-08-09:
thre are four flywoof405's, which one do you have?
```
FLWO-FLYWOOF405
FLWO-FLYWOOF405NANO
FLWO-FLYWOOF405PRO
FLWO-FLYWOOF405S_AIO
```

Maxluli, 2023-08-09:
Hey, thanks for the response. I double-checked on the flywoo website, and it appears that I have a F405PRO. The auto-detect feature did not select it, but after manually selecting the F405 the barometer acc and gyro are detected !

closed 2023-08-09 by Maxluli