betaflight/unified-targets #1115: JHEMCU F722 BT DUAL: Missing Blackbox, and Baro

pukkita, 2023-08-15:
After flashing 4.4.2 (target: JHEF7DUAL) all is apparently fine, but seems to miss drivers for:

1.- Blackbox
2.- Barometer

Maybe more, got this second hand and don't have any manuals so resorted to some google-fu: https://www.unmannedtechshop.co.uk/product/jhe-f7-bt-dual-gyro-dji-flight-controller/

PCB do seem to match visually.

Flashed to BF 4.4.2 Core and submitted report, Id: "ca83a25e-8f24-4cee-a6cd-c94e69e36340"

haslinghuis, 2023-08-15:
Core build does not reveal baro or flash drivers.

```
# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=51degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3792, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 2 locked dma shared
GYRO=MPU6000, ACC=MPU6000
OSD: MAX7456 (30 x 13)
BUILD KEY: 558ee90bd2559c6a672e6b842daf4097 (4.4.2)
System Uptime: 22 seconds, Current Time: 2023-08-15T09:45:32.210+00:00
CPU:24%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 10
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 15
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0
```

nerdCopter, 2023-08-15:
there are some defines listed here that you can test.  if they are external devices, please note they need defines as well.
https://hackmd.io/@nerdCopter/ByheaoLQh

haslinghuis, 2023-08-16:
[image]
[image]

Baro module on the left, flash chip on the right. Hard to tell with clone boards.

pukkita, 2023-08-18:
Ok, finally got proper driver to dissasemble it, turns out is a JHEMCU Supra F7 Dual:

https://yourfpv.co.uk/product/jhemcu-supra-f7-bt-fc-5v-8v-boothtooth-20x20mm/

Without blackbox nor baro.

Info (already dissappearing from aliexpress):

## JHEMCU Supra F7 BT FC 5V 8V Boothtooth

The latest design Supra F7BT mini Bluetooth, mini 20x20mm pitch F7 flight control, using STM32F722RET6 high-performance processor, MPU6000 gyroscope. It also supports Bluetooth wireless tuning.

Integrated OSD, built-in 5V/2A (internal power supply, receiver, LED, BUZZER and other power supply), 9V/3A BEC (camera and image transmission mode power supply, its opening and closing can be controlled by USER2), two independent BEC work separately , Do not interfere with each other, optimized circuit design, high-performance hardware, to further ensure the powerful performance of flight control.

Rich and complete interface definition, 6-channel UART. In addition, the 8P 1.0MM socket is designed to facilitate the direct input of the plug line, connect with BLHELI 32-bit ESC, and support data return (RX1).
Integrated Bluetooth 4.0, support mobile phone APP Bluetooth wireless parameter adjustment, convenient for field optimization parameters, especially novices no longer worry about the driver.

Product Information.
Model: SupraF7BT
Main control chip: STM32F722RET6216MHz
Gyroscope: MPU6000 gyroscope (SPI)
Barometer: None
OSD chip: AT7456E
Black box: None
LED programming: support
BB sound: support
Input voltage: 2-6S
I2C interface: support
BEC-1: 5V/2A (internal power supply, receiver, LED, BUZZER, etc. power supply)
BEC-2: 9V/3A (camera and image transmission mode power supply, its opening and closing can be controlled by USER2, turn on the power when selected, the shipment is fully open by default, please configure the channel control switch by yourself)
Serial control port: six channels (UART1, UART2, UART3, UART4, UART5, UART6)
Bluetooth: support (port UART4, baud rate 115200, name Supra)
Flight control firmware: Betaflight
Goal: JHEF7DUAL.HEX
Conversion of hole position and shock-absorbing ball hole position: 20*20mm, φ3mm
Flight control size: 30*27mm

Weight: 8g

[image]

---

Found this info: https://github.com/iNavFlight/inav/issues/5002 where smaller09 comments

```
#define USE_PINIO
#define USE_PINIOBOX
#define PINIO1_PIN PC14 // VTX power switcher
#define PINIO2_PIN PB09 // 2xCamera switcher
```

This is a Vista build, soldered to BEC2. I need to enable USER2

I tried to add these defines, and while Betaflight didn't warn me about anything, while checking the build there are warnings/errors:

https://build.betaflight.com/api/builds/109e59888556218f0fab5f8e70105d67/log

haslinghuis, 2023-08-18:
Add `PINIO` using Other Options.

After flashing go to CLI and enter:
```
set pinio_box = 40,41,255,255
set box_user_1_name = VTX SWITCH
set box_user_2_name = CAM SWITCH
save
```

pukkita, 2023-08-20:
USER2 working fine, thanks!!!!

Maybe it's just cosmetic, but when loading the BF backup I get right at the beggining:

```
# resetting to defaults
###ERROR IN resource: INVALID RESOURCE NAME: 'CAMERA_CONTROL'###
```

No resource by that name for sure.

Not sure if related, but while I see the bluetooth LED blinking indicating is enabled, I cannot find any "Supra" BT device on my phone.

UART4 set to MSP / 115200 in BF.

haslinghuis, 2023-08-20:
Adding Camera Control to Other Options will clear the invalid resource name.

BT setup seems ok:
[image]

closed 2023-08-20 by haslinghuis

pukkita, 2023-08-20:
Definitely! No warnings now, backup loads fine, thanks!

Speedybee app sees the SupraF7 as unknown device, it won't finish the connection and stay in "connecting SupraF7", so cannot use it, but Bluetooth light went from blinking to solid, so I guess BT is indeed working.