betaflight/unified-targets #1119: Gyro config on MatekF411 target is faulty (4.4.0 and 4.4.2 tested)

pfreundl, 2023-08-18:
**No gyro after upgrading from 4.3 to 4.4**

Hi, i just tried to upgrade my old Tyro79 FC (no name f411 board)[https://www.eachine.com/Eachine-Tyro79-DIY-FPV-Racing-RC-Drone-Spare-Part-Customized-F4-Flight-Controller-Integrated-OSD-p-1425.html] from BF version 4.3 to 4.4.2 and noticed that the gyro was no longer detected (gyro symbol grayed out, all config options disabled). I tried flashing back to 4.3 and the gyro worked fine again. Version 4.4.0 also does not work. The resource is connected to the same pad, so no issue there i guess. Thanks for your help!

Here is my 4.3 dump:

> 2023-08-18 @14:26:05 -- OS: Windows

2023-08-18 @14:26:05 -- Konfigurator: 10.9.0 (697943b)

2023-08-18 @14:26:07 -- Lade Github Information für Version configurator.

2023-08-18 @14:26:11 -- Serieller Port erfolgreich geöffnet mit ID: 1

2023-08-18 @14:26:11 -- MultiWii API Version: 1.44.0

2023-08-18 @14:26:11 -- Flugkontroller Info, Identifikator: BTFL, Version: 4.3.0

2023-08-18 @14:26:11 -- laufende Firmware veröffentlicht am: Jun 14 2022 00:48:04

2023-08-18 @14:26:11 -- Board: MTKS/MATEKF411(STM32F411), Version: 0

2023-08-18 @14:26:11 -- Eindeutige Geräte ID: X

2023-08-18 @14:26:11 -- Fluggerät Name: 2fourdy

2023-08-18 @14:26:11 -- Arming deaktiviert

2023-08-18 @14:26:18 -- CLI Modus erkannt

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# dump

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jun 14 2022 / 00:48:04 (229ac66) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

board_name MATEKF411
manufacturer_id MTKS

# name: 2fourdy

# resources
resource BEEPER 1 B02
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B03
resource MOTOR 6 B10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
[... 1097 lines cut ...]
set thr_expo = 0
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

nerdCopter, 2023-08-18:
This is not a bug per se, 4.4 introduces cloud-build. You will need to find out which gyro and add it to custom defines or submit a support-id after flashing "core-build".

Please see this FAQ: https://hackmd.io/@nerdCopter/ByheaoLQh

Please use Discord for support. Once you find the proper gyro or submit a support-id, you can update this ticket. Devs will fix the target-definition.

There are 3 MATEKF411 definitions, all contain MPU6000 only until updated as needed.
[image]

Also for Configurator 10.9, it is required to enable expert-mode as seen here.
development branches are optional, but 4.4.3-zulu might be better than 4.4.2
[image]

haslinghuis, 2023-08-18:
Add `ACC_SPI_ICM20602 GYRO_SPI_ICM20602` to custom defines.

pfreundl, 2023-08-18:
using expert mode on Version 4.4.2 using ACC_SPI_ICM20602 GYRO_SPI_ICM20602 worked for me. This is for an old batch around late 2018 / early 2019 if anyone else is looking for this, since there is barely any documentation / info on tyro 79 FCs.  Thank you very much for your help!

haslinghuis, 2023-08-18:
[20190122200153TYRO79EnglishManual.pdf](https://github.com/betaflight/unified-targets/files/12381287/20190122200153TYRO79EnglishManual.pdf)

closed 2023-10-07 by github-actions[bot]