betaflight/unified-targets #1120: Build server failure

GEME-Cyi, 2023-08-18:
Betaflight Configurator can't build SOLOGOODF722 firmware

Build server failure: https://build.betaflight.com/api/builds/null/SOLOGOODF722 HTTP 404

haslinghuis, 2023-08-18:
https://build.betaflight.com/api/builds/15a3983ea461149369490141626ecd2e/log

GEME-Cyi, 2023-08-18:
Betaflight Configurator can't build 4.3.2

haslinghuis, 2023-08-18:
@blckmn please check:

[image]

blckmn, 2023-08-18:
4.3.2 uses generic targets.

GEME-Cyi, 2023-08-18:
@blckmn How can I change it

haslinghuis, 2023-08-18:
@blckmn @GEME-Cyi probably means 4.4.2 not available? Unified target has been added two days ago

[image]

GEME-Cyi, 2023-08-19:
@haslinghuis Can a congfig error cause this?      #define UART3_RX_PIN B11

blckmn, 2023-08-19:
Fixed. Just re-ran the action.

closed 2023-08-19 by blckmn