betaflight/unified-targets #1146: Gyro, accellerometer not detected in Speedybee F4V3-50-Stack

SamKarthi101, 2023-09-08:
Hello Team, 

After flashing the 4.4.0 firmware to my flight controller my gyro accelerometer stopped working. The motors stopped spinning as well 

Heres the ID of the support data
*** Support data submitted *** 
Id: "e43cdeaa-94e9-461a-bf80-b29b5861beef"

# copy ID and provide to the betaflight team

Hopping to have a fix!
Have a nice day!

Thanks and regards,
Sam

nerdCopter, 2023-09-08:
are you certain you don't have the SpeedyBee**F7**V3, not the F4

SamKarthi101, 2023-09-08:
Yes I am sure of it
Heres a image of it

nerdCopter, 2023-09-08:
But are the other things working now?
[Discord](https://discord.betaflight.com/invite) is better place for conversation/support.
You can add a custom define for the Baro or send a core-build support-id if the F4 is working.
https://hackmd.io/@nerdCopter/ByheaoLQh
Maybe send a `diff` of the prior-working 4.3 or earlier  so that we are clear about things.

SamKarthi101, 2023-09-08:
> after flashing  SPEEDYBEEF4 barometer also is missing

I am unable to find the gyro and accelerometer models
Not sure if I am searching for them correctly
.

[image: image.png]

SamKarthi101, 2023-09-08:
Hello,

How can I find the gyro data for the fc that I am using?

Regards

nerdCopter, 2023-09-08:
for the records. support continued in discord.
validated F405
photos: [SpeedyBeeF405v3.zip](https://github.com/betaflight/unified-targets/files/12561892/SpeedyBeeF405v3.zip)

closed 2023-09-09 by haslinghuis

mrakus-sofomo, 2024-12-28:
The issue is still in last version. I have two boards, one of them is missing gyro/acc.

nerdCopter, 2025-01-02:
[https://hackmd.io/@nerdCopter/ByheaoLQh
Discord](https://discord.betaflight.com/) for support