betaflight/unified-targets #1149: FreeAxis 25A no Gyro, no baro, no blackbox

IKKI-FPV, 2023-09-17:
*** Dati per il supporto inviati *** 
Id: "793e2e7e-d244-4b4c-9f3b-4c2599f57804"

https://it.aliexpress.com/item/1005005844783456.html

nerdCopter, 2023-09-18:
two reports on that link about failing after update.  manufacturer may have created custom definition/hex.

https://build.betaflight.com/api/support/793e2e7e-d244-4b4c-9f3b-4c2599f57804 indeed show no gyro suing `MATEKF411`

https://build.betaflight.com/api/builds/a95d31bd5684a9366c1e0a7cf123ccb9/log shows all the gyro and blackbox chips compiled. 

MATEKF411 defines only the mpu6000 by default: https://github.com/betaflight/config/blob/master/configs/MATEKF411/config.h

do you have a `dump` from the original firmware saved?

IKKI-FPV, 2023-09-18:
Hello and thanks for the reply.  Unfortunately no, I stupidly didn't save the dump before updating.  I'm asking the seller for a copy with little success

closed 2023-10-26 by github-actions[bot]