betaflight/unified-targets #1161: SYNERGYF4 is missing Blackbox

leapole, 2023-09-29:
### Describe the bug

SYNERGYF4 is missing Blackbox in 4.4.2

It was working in 4.3

### To Reproduce

Flash 4.4.2, Go to Blackbox tab,  Read error message

### Expected behavior

Blackbox is part of the firmware

### Support ID

```txt
Id: "d14af7c0-1b70-42bd-b803-b1d5d015dde3"
```

### Flight controller

SYNERGYF4

### Other components

TBS items

### Add any other context about the problem that you think might be relevant here

No Blackbox vars in the cli.

Its present in 4.3. 

There is a 16MB chip onboard

nerdCopter, 2023-10-30:
@leapole , did you find the correct blackbox chip "custom define"? else we need a support-id from a core-build.
How to: https://hackmd.io/@nerdCopter/ByheaoLQh

leapole, 2023-11-06:
> How to: ***@***.***/ByheaoLQh

its working in 4.5

PR #1186 (2023-11-15): Update KLEE-SYNERGYF4.config
Fixes #1161
commit cb6ad139 (2023-11-15): Update KLEE-SYNERGYF4.config (#1186)

Fixes #1161
closed 2023-11-15 by haslinghuis