betaflight/unified-targets #1163: GEP RC GEP-F722-35A AIO lost Gyro and possibly other ports after flashing BF fireware 4.40 and 4.4.2

upyopyo, 2023-10-06:
### Describe the bug

Hi, I tried to flash BF fireware from the original version to 4.4.0 and 4.4.2  in order to use newer generation's GPS. And after flashing with sucessful update, Gyro is not detected anymore. (Sorry, I forgot the original BF version, possibly 4.2.x). Before flashing fireware, my drone is working fine.

At the beginning, the firmeware uploading was frozen at initializing step. Then I ran  ImpulseRC_Driver_Fixer.exe. Althoguh it finished with error message, now I can upload the BF firewares. Then I tried to flash with 4.4.0 and 4.4.2 and the both don't  detect Gyro. 

Next I found this link and a magic line for custom defines:
https://github.com/betaflight/betaflight/issues/12138
GYRO_SPI_MPU6000 ACC_SPI_MPU6000 

However, it still did not fix my Gyro detection, and now I have no idea how to fix this issue. 
Actually I found other issues. Airunit does not output video, and receiver does not receive signal although UARTs set properly as same as before flashing. Thus, I am thinking the problem is not only Gyro, but also other components, too. 

Maybe, I am missing very critical process. Please let me know how to fix all, thanks. 
These attached screenshots are after flashing frimewares. 

[image]
[image]
[image]
[image]

### To Reproduce

rerun to flash fireware updatings.

### Expected behavior

I am expecting to see detect Gyro and 

### Support ID

```txt
*** Support data submitted *** 
Id: "2bf4e59c-4bdd-44d7-b9cc-cd98510d3235"
```

### Flight controller

GEP-F722-35A AIO

### Other components

receiver Futaba
VTX DJI airunit
GPS HGLRC M100_MINI GPS
buzzer gepRC superBuzzer

nerdCopter, 2023-10-06:
10.9 has a bug that required "expert-mode" enabled. 
[image]

or you can use the 10.10-RC1 https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC1

upyopyo, 2023-10-06:
A
>>you will need firmware GEPRC_F722_AIO
Yes, I thought too, but there is no exact F722_AIO, and "_BT_" indicates a stack board item.  Mine is a just singal board.

B
>> For gyro all drivers are included in CORE build. 
umm, if you say so, , did I misstake a critical procedure after flashing firmware? At least, I hit  'connect' button' and selected 'apply custom Defaults' button. Then should it be all good, right? 

C
>>  10.9 has a bug that requires "expert-mode" enabled.
I tried to do the flash firmware 4.4.2 with 'Enable Expert Mode' ON, but it's the same story, not working. Do you imply to try development versions like pre-release?

additional questions to double check:
D
Do I need to install such Zadig and STM things? Because I can already do the flash Firmware in BF by USB connection, I don't need to do it, right? (Or this could be problem??)

E
There is no AIO version in the list of BF. Should I get a firmeware from official GEP RC website? (I don't think they have the one.)

[image]
[image]
[image]

nerdCopter, 2023-10-06:
you forgot the `GYRO_SPI_MPU6000 ACC_SPI_MPU6000` defines
[image]

upyopyo, 2023-10-06:
I tried the falshing with export mode and `GYRO_SPI_MPU6000 ACC_SPI_MPU6000`
However, it did not work. Critically Other UARTs also have not working neither since flashing. For instance, a receiver does not work although its power light is up. .

[image]

haslinghuis, 2023-10-06:
@blckmn please have a look.

 [GEPRCF_722_AIO](https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRC_F722_AIO.config) is not available in cloud build. Same for [4.5 config](https://github.com/betaflight/config/tree/master/configs/GEPRC_F722_AIO)

haslinghuis, 2023-10-06:
[image]

upyopyo, 2023-10-07:
Great, now All suncion are working, thank you so much!!!!

One last thing is about OSD's GPS satelites info. 
WTF cannot get satelite's numbers and home direction although Original OSD shows those values. I guess this issue is from new BF version 4.4. Can you fix this? or should I ask to WTF team? 

[image]
[image]

nerdCopter, 2023-10-07:
Discord for "support"; Github for bug-tracking.

upyopyo, 2023-10-10:
for now, I got solution for my first question, then clsoing this ticket. Thanks for the supports!

closed 2023-10-10 by upyopyo