betaflight/unified-targets #1165: f4 1-2s no gyro

budgieda, 2023-10-19:
### Describe the bug

Id: "06a0f8cf-b167-421b-913c-fea65f69feee"

### To Reproduce

Id: "06a0f8cf-b167-421b-913c-fea65f69feee

### Expected behavior

Id: "06a0f8cf-b167-421b-913c-fea65f69feee

### Support ID

```txt
Id: "06a0f8cf-b167-421b-913c-fea65f69feee"
```

### Flight controller

lite radio 2se

budgieda, 2023-10-20:
where to find the CLI for this model and the board firmware?  thx

haslinghuis, 2023-10-20:
You did not provide any info about brand or board. Please provide a product link from the manufacturer / web store.

budgieda, 2023-10-22:
Betafpv f4 aio 1s frsky fcc

haslinghuis, 2023-10-22:
Probably needs BETAFPVF411RX

budgieda, 2023-10-22:
I found the cli and it is throwing me errors in the console in red.

nerdCopter, 2023-10-23:
anything 4.3 and prior upgraded to 4.4 and later will have different naming for some parameters or lack of parameters if a feature was not selected in the cloud-build options.  
Discord for support, Github for bugs.  -- this is a support issue, not a bug.
https://hackmd.io/@nerdCopter/ByheaoLQh

closed 2023-10-24 by haslinghuis