betaflight/unified-targets #1167: Barometer and SD Card black box not recognized on CineApe25

Denethor85, 2023-10-22:
### Describe the bug

In BF 4.4 Cloud Build on DarwinFPV CineApe 25 the barometer do not work. It work On Core Only.... Even if I insert an SD Card for the blackbox this is not recognized and It  signals me a critical error in red.  I don't know if the cause of this could be my SD card or whatever reason... I haven't tried it in Core Only

### To Reproduce

Flash 4.4 Cloud Building

### Expected behavior

The barometer and sd-card working

### Support ID

```txt
7ec25c51-c75f-4ea3-bb87-0d93b4cb6de5
```

### Flight controller

Stock FC

MR444, 2024-01-08:
> Please follow https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

Hi, I updated to 4.4.2 and tried everything again. Barometric sensor and SD card still don't work with DarwinF411.

I flashed as you suggested and my ID is 
*** Support data submitted *** 
Id: "9059edcb-838e-4fb5-bef6-5ca929d47ecc"

Sorry I didn't find out how to do that on Discord. What can I try next?

haslinghuis, 2024-01-08:
@MR444 Please try core build to find out which driver the baro needs.

MR444, 2024-01-08:
@haslinghuis Didn't my last attempt work with Id: "9059edcb-838e-4fb5-bef6-5ca929d47ecc"? That should be a core build

haslinghuis, 2024-01-08:
Seems the board does not have a baro as it does not show up in core build. Can you try 4.5RC2 also?

MR444, 2024-01-08:
Here is the ID with the new 4.5RC2
*** Support data submitted *** 
Id: "0ddf092a-db42-4454-9c1d-ae280ba7d2ab"

haslinghuis, 2024-01-08:
Try setting `set i2c1_clockspeed_khz = 800`

MR444, 2024-01-08:
With which firmware? 4.4.3 or 4.5 RC2?

And set i2c1_clockspeed_khz = 800 works also with non-core and baro_spi_device = 3 only in core?

haslinghuis, 2024-01-08:
baro spi setting would require other drivers

MR444, 2024-01-08:
Sorry what does that mean for me? Please write it for stupid people, with which firmware in which mode (core or not) I could try these changes.

haslinghuis, 2024-01-08:
Only for testing - use 4.5 RC2 core build and try

```
set i2c1_clockspeed_khz = 800
save
```

```
try set baro_spi_device = 3
save
```

Try both with USB only and with USB and lipo
Also check if barometer is enabled in configurator on configuration tab.

If it does not work you will have to contact manufacturer as we don't have direct contact and the manufacturer has supplied the configuration for this board.

MR444, 2024-01-08:
Okay I tried both now, first the clockspeed change and then the baro-spi change but didn't change anything. 

Although activated in the configuration tab, I do not get any info in the sensors tab.

I already contacted the manufacturer some days ago. I hope they will reply soon.

At least the SD card is working now! I think the trick was to flash the firmware while the SD card is inserted.

This is how it looks like with 4.5 RC2 core build and your suggested changes:
```
# dump

# version
# Betaflight / STM32F411 (S411) 4.5.0 Jan  8 2024 / 13:28:32 (f1cbd83f0) MSP API: 1.46
# config rev: ed9a9aa

# start the command batch
batch start

board_name DARWINF411
manufacturer_id DAFP

# name: -

# resources
resource BEEPER 1 B02
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B03
resource MOTOR 6 B10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
[... 868 lines cut ...]

# rateprofile 0
set rateprofile_name = -
set thr_mid = 50
set thr_expo = 0
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end
```

haslinghuis, 2024-01-08:
Sorry - forgot to mention also have to set

```
set baro_bustype = SPI
```

when trying SPI bus (need drivers in core build)

MR444, 2024-01-08:
Nope still no data. Neither in the Sensors tab nor in the OSD. Also not listed in the status command:

`# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.30V, Core temp=44degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3644, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6500, ACC=MPU6500
OSD: MAX7456 (30 x 13)
BUILD KEY: 0d63c444f7bec6d628b558975268c11f (4.5.0-RC2)
System Uptime: 129 seconds, Current Time: 2024-01-08T21:40:47.720+00:00
CPU:61%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1467 * 0.01V (4S battery - OK)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP`

haslinghuis, 2024-01-08:
Did you try 4.3.2  to check if baro is working?

MR444, 2024-01-08:
Yes this was the really first thing what I did. You think it could be a hardware failure?

MR444, 2024-01-15:
Hi, I finally got response from the manufacturer that the FC does not support barometer although it says on their website that is has 
_DarwinFPV 3-6S F411 3-4S 30A stack has **built-in barometer** and SD card slot, and reserves rich interfaces resource, which meet the needs of lightweight flight._

Sorry for wasting your time, they could have told us earlier and could have correct data on their website.

haslinghuis, 2024-01-15:
@MR444 thanks for reporting back!

Denethor85, 2024-01-15:
Attention that with Betaflight released as stock in the FC both the barometer and the SD card work if I remember correctly... I also had this problem in fact I had opened the post but I don't remember how I solved it.... I'll have to check again but at least the SD card for blackbox is sure to work... This in CineApe25 stock FC

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

MR444, 2024-01-15:
I cannot confirm that. With the stock firmware, both SD and barometer did not work. I could only make the SD card work when I flashed latest stable BF with SD card inserted.

Denethor85, 2024-01-21:
This Is my CineApe with default 4.3.2 Betaflight from DarwinFPV...

Baro work and Blackbox with Sd Card work....

[image]

[image]

MR444, 2024-01-22:
Strange, mine doesn't work with stock FW. Maybe they changed something or I really got a faulty board.
But then the message from the support is wrong.

Denethor85, 2024-01-22:
With Betaflight 4.3.2 try with this command on cli

Barometer
```
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE

set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 1
set baro_i2c_address = 0
set baro_hardware = AUTO
set baro_tab_size = 21
set baro_noise_lpf = 600
set baro_cf_vel = 985
```

SD Card and Blackbox
```
resource SDCARD_CS 1 A00
resource SDCARD_DETECT 1 NONE
resource FLASH_CS 1 NONE

set blackbox_sample_rate = 1/2
set blackbox_device = SDCARD
set blackbox_disable_pids = OFF
set blackbox_disable_rc = OFF
set blackbox_disable_setpoint = OFF
set blackbox_disable_bat = OFF
set blackbox_disable_mag = OFF
set blackbox_disable_alt = OFF
set blackbox_disable_rssi = OFF
set blackbox_disable_gyro = OFF
set blackbox_disable_acc = OFF
set blackbox_disable_debug = OFF
set blackbox_disable_motors = OFF
set blackbox_disable_gps = OFF
set blackbox_mode = NORMAL

set sdcard_detect_inverted = OFF
set sdcard_mode = SPI
set sdcard_spi_bus = 2

set flash_spi_bus = 0
```

Attention, if I remember correctly the SD Cards must be a maximum of 32Gb, not all models are recognized and above all it must be formatted in FAT but the "SD Card Formatter" program must be used to make it compatible. 

Kindly let me know if everything works again after this. However all this in version 4.3.2. I remember that when I tried version 4.4 I also had some problems, in fact I opened this report but I don't remember how it ended but I certainly now have version 4.3.2 on CineApe25

MR444, 2024-01-24:
Thank for your help. 

I cannot confirm yet if it works because I am not sure if I want to flash back to 4.3.2 when I am quite happy with 4.4 at the moment or if I will anway switch to another stack with more UARTs.

Denethor85, 2024-01-24:
> Thank for your help. 
> 
> I cannot confirm yet if it works because I am not sure if I want to flash back to 4.3.2 when I am quite happy with 4.4 at the moment or if I will anway switch to another stack with more UARTs.

Maybe you can this on 4.4 but you must flash with appropiate driver

r2df, 2024-01-28:
My blackbox seems to be working but the logs are "corrupted" (only some data points are recorded, so it is useless). Sometimes it doesn't even save a log file.
I am using BF 4.4.2, and a SanDisk Ultra 8GB. It is also weird that it sais 3.4GB not available, but the SD is 8GB:
[image]

What could I try to have well recorded balckbox logs?

Regarding baro, I can confirm that my cineape25 has no baro soldered in the board.

haslinghuis, 2024-01-28:
SDCard and OSD are sharing same SPI bus so blackbox will never be optimal depending on [number of] OSD elements used.

Hochmueller, 2024-03-31:
I could only get the logging to work if I disable the OSD in Betaflight. (not with a mode but disable the whole feature). Even with this I can only get 500Hz logs running.