betaflight/unified-targets #1182: Gyro locked

balauru327, 2023-11-01:
I have tried an firmware update and since then I can not bind my drone to my Jumper T Lite v2 and my gyro is locked. 
Id: "a51065ef-36dd-43ee-b417-49315e2615d2"
Happymodel Mobula 7 HD 1s
CRAZYBEEF4SX1280
Betaflight 4.3.0 firmware
Betaflight Configurator 10.9.0 (697943b)
Stock Mobula 7 HD 1s
Thanks.

nerdCopter, 2023-11-02:
ELRS?
for sx1280 (SPI) flight controllers, 
betaflight 4.3 will use ELRS version 2
whereas betaflight 4.4 and later will use ELRS version 3.
Your transmitter must match ELRS versions.

i'm unsure about the gyro-lock, someone else will have to answer.

Betflight Discord is a better "support" avenue. https://discord.betaflight.com/
Github is more for source-code development and bug-reporting

balauru327, 2023-11-08:
Thanks. I know that the BF firmwares above 4.4 needs ELRS v3 so I tried all the versions 4.3.x.
I will try Discord. 
Thanks.

closed 2023-12-17 by github-actions[bot]