betaflight/unified-targets #1235: Flash Hobby F405 no Gyro or accel after update to 4.5.0

Username-unavailable1, 2024-01-07:
Id: "adc56e2b-d25d-4d1a-b38a-17b3a47edba4"
When flashing to 4.5.0 I have No Gyro or accel  Flashhobby F405 Board (https://www.banggood.com/30_5x30_5mm-Flashhobby-F405-MPU6000-F4-OSD-Blackbox-Flight-Controller-with-5V-9V-BEC-Output-and-60A-BL_S-4in1-ESC-3-6S-Stakc-for-RC-FPV-Racing-Drone-p-1994714.html?rmmds=myorder&cur_warehouse=CN) Flashing to Flashhobby F405 target.
Reproduce: Flash F405 Board with 4.5.0 with other options: GPS LED strip OSD (SD) VTX. CRSF Radio protocol, DSHOT motor protocol

nerdCopter, 2024-01-08:
https://hackmd.io/@nerdCopter/ByheaoLQh

Username-unavailable1, 2024-01-09:
> the above supportID is not a core-build.
Oh really? I swore that it was flashed to core but just in case here is one I made sure was core 
Id: "a9de0523-9a36-4de7-b321-2bc424bad552"
> 
> https://hackmd.io/@nerdCopter/ByheaoLQh using the above FAQ, please report which gyro works.
I have tried this, however the configurator comes up with an error like this for every one 

`In file included from ./src/main/platform.h:32,
                 from ./src/main/drivers/inverter.c:24:
./src/config/configs/FLASHHOBBYF405/config.h:30: error: "USE_ACC_SPI_MPU6000" redefined [-Werror]
   30 | #define USE_ACC_SPI_MPU6000
      | 
<command-line>: note: this is the location of the previous definition
In file included from ./src/main/platform.h:32,
                 from ./src/main/drivers/inverter.c:24:
./src/config/configs/FLASHHOBBYF405/config.h:34: error: "USE_GYRO_SPI_MPU6000" redefined [-Werror]
   34 | #define USE_GYRO_SPI_MPU6000`
> 
> pasting entire 4.4 or prior diffs not recommended as there may be differing parameter names. ignore errors, or choose wisely which settings to paste.
Oh ok, that is just what made it work on 4.4.3 (if I didn't put the presets in I would have the same issue of no gyro accel and baro). Is there documentation I can read up on about this?

Here is a core flash of 4.4.3 WITH backup presets loaded where it states that the accel/gyro is the MPU6000 and baro is BMP280: 
Id: "ba442036-187e-4cfb-a63c-1da7dee68ab2"
If I don't put the backup preset that I saved before I updated to 4.4.3 (when I got the board) then nothing shows up (Accel,gyro,baro) just like when I update to 4.5.0-RC2

I am using the new configurator 10.10.0RC-2 and it doesn't seem to have solved the issue if that's worth noting.
Also I don't have esc passthrough on 4.5.0 not sure if it's related or not

nerdCopter, 2024-01-10:
okay, weird. your new supportid with CORE build still has no gyro too :frowning_face: 
https://build.betaflight.com/api/support/a9de0523-9a36-4de7-b321-2bc424bad552
https://build.betaflight.com/api/builds/d8b7632a201403bc9d2b9822c9e31ff9/json

regarding errors, if you define one that already exists, it will error.

core log show all gyro's built https://build.betaflight.com/api/builds/d8b7632a201403bc9d2b9822c9e31ff9/log
[image]
Sorry, don't know how to help. maybe FC broken.

nerdCopter, 2024-01-10:
be sure to full-chip erase.
connect without pasting diff/dump. still bad or good? if good, do not paste entire dump.

Username-unavailable1, 2024-01-11:
> be sure to full-chip erase. connect without pasting diff/dump. still bad or good? if good, do not paste entire dump.

Yea I full chip erase every time, I was kinda hoping others might have an issue and chime in but you probably won't until 4.5.0 comes out of release candidate, Thanks anyways I'll keep digging lol

Username-unavailable1, 2024-01-14:
> If you have a working hex / dump then we can look at the resource changes

Hey how exactly do you make a hex of your current configuration? I've looked all over and I can't find anything to tell me how to make one, only how to install one. Thanks!

haslinghuis, 2024-01-14:
You seem to have a non standard board as the configuration should work. Ask manufacturer if there are changes between revisions of this board. Assume this is not a clone board.

Username-unavailable1, 2024-01-19:
> You seem to have a non standard board as the configuration should work. Ask manufacturer if there are changes between revisions of this board. Assume this is not a clone board.

No response from the seller... lol how do you tell if its a clone? the link in my first comment is where I bought it from. Thanks!

guyilherme, 2024-02-03:
> Id: "adc56e2b-d25d-4d1a-b38a-17b3a47edba4" When flashing to 4.5.0 I have No Gyro or accel Flashhobby F405 Board [...]

I have the same board, purchased from the same store and the only versions I can connect to are 4.3.1 and 4.3.2, however, with the gyro error you described. Can you pass the .hex that you used and that worked on your board?

haslinghuis, 2024-02-03:
In CLI tab run `dump hardware` and use three backticks ( ` ) to create a [fenced] block.

```
and put output here
```

https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

guyilherme, 2024-02-03:
> In CLI tab run `dump hardware` and use three backticks ( ` ) to create a [fenced] block.
> 
> ```
> and put output here
> ```
> 
> https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

I don't know if your response was directly to me, but here is the dump of my FC.

```
# 

# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.3 Nov 14 2023 / 16:02:47 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name FLASHHOBBYF405
manufacturer_id FHRC

# resources
resource BEEPER 1 B02
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B05
resource MOTOR 4 B04
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 C08
resource MOTOR 8 C09
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
[... 205 lines cut ...]
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

# 
```

haslinghuis, 2024-02-03:
This dump is working on 4.4.3 ?

haslinghuis, 2024-02-03:
Oh, try enabling dshot_bitbang

guyilherme, 2024-02-03:
> Oh, try enabling dshot_bitbang

How exactly do I enable this option?

haslinghuis, 2024-02-03:
```
set dshot_bitbang = ON
save
```

guyilherme, 2024-02-04:
> ```
> set dshot_bitbang = ON
> save
> ```

Same problem. 

```
# 

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.31V, Core temp=45degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3640, max available: 16384
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: MSP (30 x 13)
BUILD KEY: 5e6c7dd1e877ed7ccc28c8ae7dac0676 (4.4.3)
System Uptime: 41 seconds, Current Time: 2024-02-04T00:29:21.423+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 329 * 0.01V (1S battery - WARNING)
I2C Errors: 0
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# 
```

haslinghuis, 2024-02-04:
Then we need the working dump to check what is wrong.

Username-unavailable1, 2024-02-04:
> Then we need the working dump to check what is wrong.
**Here is working dump on 4.4.3** 
`# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.3 Nov 23 2023 / 22:40:22 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name FLASHHOBBYF405
manufacturer_id FHRC

# resources
resource BEEPER 1 C13
resource MOTOR 1 A08
resource MOTOR 2 A09
resource MOTOR 3 A10
resource MOTOR 4 C08
resource MOTOR 5 C09
resource MOTOR 6 B00
resource MOTOR 7 B01
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B02
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 C05
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
[... 557 lines cut ...]
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = DEFAULT
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end
`
And for the record, dshot bitbang does nothing for me aswell

Username-unavailable1, 2024-02-04:
ok so just ran these through notepad++ and wow... It's like they are 2 completely different boards, gonna try manually changing and report back

Username-unavailable1, 2024-02-04:
OK, so here is the updated hardware dump after I went through and corrected all the pins (4.5.0RC-2)
```# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.5.0 Feb  4 2024 / 17:19:29 (f1cbd83f0) MSP API: 1.46
# config rev: 43beb50

# start the command batch
batch start

board_name FLASHHOBBYF405
manufacturer_id FHRC

# resources
resource BEEPER 1 C13
resource MOTOR 1 A08
resource MOTOR 2 A09
resource MOTOR 3 A10
resource MOTOR 4 C08
resource MOTOR 5 C09
resource MOTOR 6 B00
resource MOTOR 7 B01
resource MOTOR 8 NONE
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B02
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 C05
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource SOFTSERIAL_TX 1 NONE
resource SOFTSERIAL_TX 2 NONE
resource SOFTSERIAL_RX 1 NONE
resource SOFTSERIAL_RX 2 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 B10
resource I2C_SCL 3 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 B11
resource I2C_SDA 3 NONE
resource LED 1 C14
resource LED 2 C15
resource LED 3 NONE
resource SPI_SCK 1 A05
[... 211 lines cut ...]
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end
```
I got my BARO back but thats it still no gyro or accel...
The only thing that I noticed different was that the SPI_MISO and SPI_MOSI Pins turned into SPI_SDO and SPI_SDI pins, not quite sure what this means as google just spits out really complicated stuff about micro chips lol

haslinghuis, 2024-02-04:
| Term | Description | Deprecated |
| :--- | :--- | :--- |
| SDO | Serial Data Out | Master Out Slave In (MOSI) |
| SDI |  Serial Data In | Master In Slave Out (MISO) |

haslinghuis, 2024-02-04:
What ACC/GYRO was reported in status command in CLI while it worked?

Username-unavailable1, 2024-02-04:
> What ACC/GYRO was reported in status command in CLI while it worked?

accel/gyro is the MPU6000 and baro is BMP280

haslinghuis, 2024-02-04:
Yes MPU6000 is already included, SPI pins seems correct also.

closed 2024-03-13 by github-actions[bot]