betaflight/unified-targets #1250: TinyApe Flight Controller (BETAFPVF4SX1280) Missing Gyro after BetaFlight 4.4

spla9t-lab, 2024-02-06:
After updating to Betaflight 4.4 on the TinyApe (MCU: STM32F411CEU6 - Firmware target BETAFPVF4SX1280) the gyro is missing due to a mismatched gyro in the cloud configuration.

This can be reproduced by updating the DarwinFPV TinyApe from Betaflight 4.3 to Betaflight 4.4 and attempting to reconnect the flight controller (MCU: STM32F411CEU6 - Firmware target BETAFPVF4SX1280).

This was submitted with the Support ID: 769acffa-228d-47fb-8dbd-2240123c6033

Thanks for everything you do!

haslinghuis, 2024-02-06:
Darwin needs it's own configuration file instead of using another manufacturers work.

ot0tot, 2024-02-06:
There is already a corrected target for the Darwin FC:
https://github.com/betaflight/unified-targets/blob/master/configs/default/DAFP-DARWINF4SX1280HD.config

closed 2024-02-06 by spla9t-lab