betaflight/unified-targets #1262: OmnibusF4 FC gyro not working after update to betaflight 4.4 firmware

Mikkupk, 2024-02-29:
*** Support data submitted *** 
Id: "c86adbf7-1d48-4f75-855f-befd321c3428"

haslinghuis, 2024-02-29:
Please provide URL of the board used - as there are many clones out.

Try adding custom defines: 
`GYRO_SPI_ICM20602 ACC_SPI_ICM20602`

OR

`GYRO_SPI_ICM20689 ACC_SPI_ICM20689`

OR

do core only build and provide new support ID.

Mikkupk, 2024-02-29:
Dear Mark Haslinghuis,

Thanks for your time and consideration. I have no other details about this
board, I just removed from one of my fpv that I bought from one another
person. Here Im adding its photos, that may be can help you to recognize
some of the relevent datas. And I will do further testing using the data
that you already provided to me. I hope to hear from you soon.

Yours sincerely,
Prince

haslinghuis, 2024-02-29:
Checkout http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab

closed 2024-04-08 by github-actions[bot]