betaflight/unified-targets #1274: F4 fc gyro is disenabled

Fpvspecial, 2024-03-13:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

Fpvspecial, 2024-03-13:
*** Supportdaten übermittelt *** 
ID: "0889a431-40d5-4ed7-8b56-0d4e4c786da3"

nerdCopter, 2024-03-13:
https://hackmd.io/@nerdCopter/ByheaoLQh

MATEKF411 defines only MPU6000 at this time.
Is your flight-controller a clone? There may be a specific target for your clone FC.

There are previous issue-tickets referencing `MATEKF411` as well: https://github.com/betaflight/unified-targets/issues?q=is%3Aissue+MATEKF411

A quick fix should be to include `ACCGYRO_BMI270` in custom defines.

haslinghuis, 2024-03-13:
What board is this? The manufacturer should provide it's own configuration and not use original ones. This won't be included in the original MATEKF411 target.

closed 2024-04-20 by github-actions[bot]