betaflight/unified-targets #1315: Instantly throttles up to 100% on arming

JaAves, 2024-05-07:
As described in the title all motors go crazy after arming without any inputs given from Tx.
Just bought 2 new Speedybee Stacks F7 V3. Same Problem on both Stacks.
I can switch one Fc or the ESC and still the problem behaves the same.
Tried BF 4.5 and 4.3 after the failure. Same problem on both versions.
Stack is pointing forward and the 3d model moves like expected. All motors work perfectly in motor tab. elrs is soldered on UART2 like described on the stacks manual.

ID: "57592ab0-2266-4842-8f9d-c5dd79eeb9d7"

Thanks for your time,
JaAves

nerdCopter, 2024-05-07:
- make sure you have either a "stable" or maybe "latest" ESC firmware on your ESC.
- make sure bi-directional dShot on both ESC firmware and Betaflight is the same, either `off`/`off` or `on`/`on` depending on your need.
- maybe lower your Damping slider to 0.7 until you figure out a root cause.

inviting others to help, but also this should be a Discord discussion as it it unlikely a bug.

JaAves, 2024-05-07:
Thanks for the fast reply!

ESC Firmware is the newest: 32.10 with PWM Frequenzy High by RPM and Low with 16khz. I think this puts bi-direction to On in BLHeliSuitw32 ?!
I put  Airmode to Off, bi-directional Dshot on, DShot 300 and dinamic idle to 10

OMG cant believe it... i just put the masterMultiplayer to 0.5 and it starts to hover... its not my first SpeedybeeF7V3 and couldnt believe that it would freak out with even slightly lowered stock PIDS... Thanks for your Help!!

closed 2024-05-07 by haslinghuis

nerdCopter, 2024-05-07:
- low D-term is just a circumvention. need to tune both Betaflight and ESC configs properly.
- bi-dir dShot is independent of PWM frequency.

https://discord.betaflight.com/