betaflight/unified-targets #1325: No gyro after flashing BetaflightF4 target to SUPER F405 HD ELRS AIO

RCFPV, 2024-05-18:
Id: "f35f1c65-e35f-4eb9-936c-fc2ba6f1e5a3" 

on CADDXFPV website it says that it is ICM42688 gyro

haslinghuis, 2024-05-18:
Link ?

RCFPV, 2024-05-18:
Hi Mark sorry this is my first time reporting an issue.
It works when I flashed betaflight with custom defines GYRO_SPI_ICM42688P
ACC_SPI_ICM42688P. So looks like gyro drivers are missing in BETAFLIGHTF4
target for SUPER F405HD ELRS AIO

thanks

haslinghuis, 2024-05-18:
[link](https://www.happymodel.cn/index.php/2024/01/05/super-f405hd-elrs-aio-3in1-flight-controller-built-in-uart-2-4g-elrs-and-20a-esc-for-hd-whoops-25-5x25-5mm-mount-hole/)

RCFPV, 2024-05-18:
Ehh. My bad. Sorry. I should've checked it there first :/

Thanks for your time.

haslinghuis, 2024-05-18:
Because it is our policy to discourage the reuse of OEM targets, we will not add drivers for the sake of targets used by other companies

closed 2024-05-18 by haslinghuis