betaflight/unified-targets #1327: I am new so I don’t know how to download butterfly configuration. I need help with that.

MehalAlamim, 2024-05-25:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

KarateBrot, 2024-05-27:
You can go here for the releases: https://github.com/betaflight/betaflight-configurator/releases

Or for the browser version just go to: https://app.betaflight.com/

haslinghuis, 2024-05-27:
We cannot help with butterfly as this is Betaflight.

Otherwise please use [discord](https://discord.com/invite/gV4XSq3fus) for further Betaflight support

closed 2024-05-27 by haslinghuis