betaflight/unified-targets #1343: Problem with recognizing gyro, barometer and accelerometer in Bf at the Iflight Blitz 745 Flightcontroller

Ultima51, 2024-08-31:
Hey 
just need a bit uf help from some experts here. My problem is that Iflight Blitz F745 has no gyro in Betaflight (10.10.0) I tried flashing a older version (just in case the new has some errors) but no results. The simulated drone in Bf on the first tab isnt moving at all when i plug in the fc and move it around. I tried contactig Iflight but no response yet. Would be grate if someone could help.
Id: f2a8bd0c-1a3c-42c5-9e48-014af18dca1b

haslinghuis, 2024-08-31:
- Perhaps try `IFLIGHT_F745_AIO_V2`
- If that does not work flash with Core only and provide support ID again.

Ultima51, 2024-08-31:
e26e8d32-f6c4-40a7-bdff-9a38f2892a64 thats the new ID 
And flashing with the Iflight_F745_AIO_V2 wont work either

haslinghuis, 2024-08-31:
Even with core build it does not enable GYRO/ACC:

```
Devices detected: SPI:0, I2C:0
```

- At this point ask manufacturer for support.
- Provide any information about the flight controller (link).
- Are you sure this is the right target ?

Ultima51, 2024-10-01:
Problem solved. it was the wrong target.

closed 2024-10-01 by haslinghuis