betaflight/unified-targets #1345: No gyro/acc detected after updating Stable F411 FC to BF 4.5

willrawls, 2024-09-24:
BF CLI Id: "39f33bd2-bfce-4fc5-bb71-61e56f235328"

Works fine on BF 4.3 but no gyro/acc is detected on BF 4.4 and up.

Firmware Target: MATEKF411

Link to FC manual (https://www.geprc.com/download/en/GEPRC-GEP-STABLE-F411-Manual-EN-V1_0.pdf)

haslinghuis, 2024-09-24:
Links shows it is using MPU6000 which is in the target already. Flash with Core Only and provide a support ID again.
Suspect `GYRO_SPI_ICM20602 ACC_SPI_ICM20602` in custom defines while flashing would work.

willrawls, 2024-09-24:
Core only bf 4.5 CLI Id: "4a2c24be-4062-4aae-8f76-bf29a32bba41"

haslinghuis, 2024-09-24:
Hmm, still does not show up gyro / acc. 
Flash working 4.3x and submit support ID to see if we can find any difference

willrawls, 2024-09-24:
BF 4.3.2 with working gyro/acc support Id: "b4733470-596d-46b6-90f1-600f68c58308"

haslinghuis, 2024-09-24:
Hmmm, GYRO=ICM42688P, ACC=ICM42688P

Add `GYRO_SPI_ICM42688 ACC_SPI_ICM42688` to custom defines while flashing

willrawls, 2024-09-24:
That worked, will do test flight. Thank you for your help!

ot0tot, 2024-09-24:
What happens if you flash the correct target? 😆 
You should be flashing `GEPRCF411`.

closed 2024-09-25 by willrawls