betaflight/unified-targets #1357: Sensors not detected after firmware update to 4.5.1

Be133, 2024-11-08:
## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
Everything looked fine when I have recieved FC from the manufacturer 
Betaflight configuration could not recognize board when clicking Auto-detect, 
I have flashed it to multiple variations of OMNIBUS, MatekF405 targets, but i could not find the correct target where all sensors are detected.
Now I am trying   betaflight_4.5.1_STM32F405.hex 

2. Include ways to reproduce the problem;
Get this FC
https://www.aliexpress.com/item/1005006149035804.html?spm=a2g0o.order_list.order_list_main.56.4f1e1802Yg0n1C#nav-specification
Flash firmware without creating a backup

3. Provide as much information as possible about your hardware and software, including:
F4 V3S PLUS FC
Item name: F4 V3 PLUS Flight Controller
Size: 36x36mm
Mounting holes: 30.5x30.5mm
STM32 F405 MCU
Dshot support
Drag and Drop OSD configured via Betaflight
Gyro:BMI270
SBUS / PPM AND Spketrum DSMX ports
MicroSD card blackbox
Baro (BMP280)
BEC: 5V3A/9V3A
On-board video filters (only 5V to VTX and camera)
Please note: The F4 V3S Plus Flight Controller only support Betaflight software, INAV firmware is not supported!

*** Support data submitted *** 
Id: "66e5b6b1-e8cb-4941-a2d8-002fb25a011a"

nerdCopter, 2024-11-08:
- Did you save a backup before or when flashing?
- it is probably a clone. original definitions may not support clones. may need to add custom defines to make it work. (`ACCGYRO_BMI270`)
- https://hackmd.io/@nerdCopter/ByheaoLQh

haslinghuis, 2024-11-08:
Seems this board has no supported target.

Be133, 2024-11-08:
I have tried adding custom define USE_ACCGYRO_BMI270, I will try again after removing 'USE_'.

haslinghuis, 2024-11-08:
Yeah custom defines uses `ACCGYRO_BMI270` instead of `USE_ACCGYRO_BMI270`

Be133, 2024-11-08:
Gyro still not detected 
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=43degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: UNCONFIGURED, size: 3790, max available: 16384
Devices detected: SPI:2, I2C:0
Gyros detected: locked
, BARO=BMP280
OSD: MSP (53 x 20)
BUILD KEY: d342e23cc2c6e7a5c46bf60dab4033e9 (4.5.1)
System Uptime: 29 seconds, Current Time: 2024-11-08T22:12:29.351+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: None inserted
GPS: NOT ENABLED
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

haslinghuis, 2024-11-08:
Boards have unique configurations in regard to pin assignments.

Be133, 2024-11-08:
And I cant find options that i have provided in custom define under options in setup tab
USE_SERIALRX_CRSF
USE_SERIALRX_FPORT
USE_SERIALRX_GHST
USE_SERIALRX_IBUS
USE_SERIALRX_JETIEXBUS
USE_RX_PPM
USE_SERIALRX_SBUS
USE_SERIALRX_SPEKTRUM
USE_SERIALRX_SRXL2
USE_SERIALRX_SUMD
USE_SERIALRX_SUMH
USE_SERIALRX_XBUS
USE_DSHOT
USE_PWM_OUTPUT
USE_TELEMETRY_FRSKY_HUB
USE_TELEMETRY_HOTT
USE_TELEMETRY_IBUS_EXTENDED
USE_TELEMETRY_LTM
USE_TELEMETRY_MAVLINK
USE_TELEMETRY_SMARTPORT
USE_TELEMETRY_SRXL
USE_ACRO_TRAINER
USE_BATTERY_CONTINUE
USE_DASHBOARD
USE_ESCSERIAL_SIMONK
USE_GPS
USE_LED_STRIP
USE_MAG
USE_OSD_SD
USE_OSD_HD
USE_PINIO
USE_SERVOS
USE_VTX

Be133, 2024-11-09:
Could this issue  be resolved within Betaflight configurations?
Can you give any suggestions or advice to help me correctly configure my FC?

Have i understood correctly https://hackmd.io/@nerdCopter/ByheaoLQh
```
Although it is the manufacturers' responsibility to keep targets updated, Betaflight developers can easily fix these issues upon reports of missing defines. Note: Manufacturer of Clones should create their own target.
```
Beta flight  can add a new Bord configuration for this clone board ?

nerdCopter, 2024-11-11:
> And I cant find options that i have provided in custom define under options in setup tab 

if it's not in one of the selections, you type them in
[image]

> Have i understood correctly 

betaflight devs will only fix or add defines on non-clone definitions.

> Beta flight can add a new Bord configuration for this clone board ?

I personally do not know that board and do not see a factory `dump` anywhere. At minimal such is needed. manufacturer should crate a config.h definition file and submit it for inclusion.  
You should ask the manufacturer for the correct firmware to use and for a factory configuration `dump`.

closed 2024-12-20 by github-actions[bot]